PX4/PX4-Autopilot #14997: Flips/crashes in SITL

julianoes, 2020-05-29:
I stumbled upon weird failures in SITL with lockstep at 20x where the vehicle flips during RTL.

I've seen this at least twice now, and it's potentially related to #14835.

https://logs.px4.io/3d?log=4d4c672d-2330-4ab6-9c1d-215c09f4bbd7

julianoes, 2020-05-29:
In the logfiles I see long drops where it supposedly doesn't log at all. After a first one it doesn't crash but after the second one it does.

MaEtUgR, 2020-05-29:
In 3D replay it looks really strange. The drone moves in a diagonal very straight line to where it flips on the ground
[image]

MaEtUgR, 2020-05-29:
Like you wrote from the logs it looks like there are two blackouts. The first one is during descending and the second one is when it already hit the ground:
[image]

julianoes, 2020-05-29:
Right, so the dropouts could just be that the logging doesn't keep up but everything else still runs. Presumably, with lockstep there shouldn't be a case where the simulation continues if there are no more motor outputs, so said differently the CPU being busy should not matter for controls purposes.