PX4/PX4-Autopilot #15003: V1.11 Pixfalcon doesn´t talk anymore with Spektrum satellite

taileron, 2020-05-30:
after the update to v1.11 the Spektrum Satellite SPM 4648 is no longer recognized.
It only shows up one time immediately after the update procedure if PicFalcon had no powercycle.
After the first power cycle the receiver is then gone and "Manual control lost" shows up in QGC.

taileron, 2020-06-19:
it concerns not only 4648 but all serial spectrum receivers also the 5V versions like 4649 or 4650. Manual Control lost, tested on both Spekt DSM and SBUS connector.

taileron, 2020-07-11:
sys_use_io = 0 enables DSMX rc function but causes other issues. #15155 

with v1.10 it´s the opposite way DSMX receiver is only detected if sys_use_io = 1

taileron, 2020-07-23:
In 1.11 with IO the 3.3V supply voltage to the serial Spektrum DSM(X) is active simultaneously with the FCs voltage.
In V1.10 and earlier this voltage starts much later in the FC´s boot sequence.

taileron, 2020-07-26:
remove pwm_input and add rc_input in boards fmu-v3 default.cmake fixes it, then the DSM(X) sat is is always recognized sustainably with and without IO.

nrogelio, 2020-08-20:
Hi,

A PR was merged to master: #15430

You can try to write master firmware to your board and see if that solves the issue. Of course master firmware is a developer version so use it at your own risk.

Regards.

taileron, 2020-08-20:
the receiver is still not recognized with IO active and since the PR#15430 the only workaround mentioned above doesn't work anymore.

dagar, 2020-08-20:
> In 1.11 with IO the 3.3V supply voltage to the serial Spektrum DSM(X) is active simultaneously with the FCs voltage.
> In V1.10 and earlier this voltage starts much later in the FC´s boot sequence.

Thanks @taileron, I'll dig into the init changes.

closed 2020-12-11 by LorenzMeier