PX4/PX4-Autopilot #15014: PX4 v1.10.1 EKF2 GPS and VIO issues

nbelanger99, 2020-06-02:
We're trying to use an both RTK GPS for position and an Intel Realsense [T265 for VIO](https://github.com/Auterion/VIO), and we're having some cofusion.

We've been able to fuse each sensor individually with decent reliability and accuracy, but it seems there may be discrepancy between frame conventions or something. 

Here we see the same exact test, moving forward and back, then left and back, then rotating anti-clockwise with the T265 (EKF2_AID_MASK = 272), and the GPS (EKF2_AID_MASK = 1), respectively 
[image]

[image]
_______________________________________________________

We can see from this view of a simpler test- walking forward then left that there's a rotation between the two different inputs, ( and I suppose the local_position estimate aligns itself with one or the other if only using one?)
[image]
__________________________________________________________________________
If we fuse vision velocity and yaw with the "rotate EV" flag set (EKF2_AID_MASK = 336), and compare the local pose fused with T265 odom to the RTK data, it would seem as though we're on the same page:
[image]
______________________________________________________________________
However, trying to add GPS fusion these same settings (EKF2_AID_MASK = 337) result again in discrepancy:
[image]

I'm pretty confused on this, any help would be appreciated.

julianoes, 2020-06-02:
FYI @baumanta @jkflying

kamilritz, 2020-06-03:
I mostly worked with EKF_AID_MASK 321 (https://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#ekf2_extvis).
For inspecting the data, I always used the ulog. There you can see both `vehicle_visual_odometry` and `vehicle_visual_odometry_aligned`. As well as all the innovation data.