PX4/PX4-Autopilot #15022: Getting GPS and Barometer error

deepak-1530, 2020-06-03:
**Describe the bug**
I am using the firmware version v1.10.1 and facing issues with the SITL. On starting the simulator, I get the following output

pxh> INFO  [ecl/EKF] 1532000: EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [ecl/EKF] 6092000: EKF commencing GPS fusion
WARN  [ecl/EKF] 6708000: EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] 11716000: EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] 26364000: EKF baro hgt timeout - reset to GPS
WARN  [ecl/EKF] 26364000: EKF gps hgt timeout - reset to GPS

PX4/Firmware (stable 1.10.1) + MAVSDK (0.26.0) + MAVSDK-Python (0.7.0).

**To Reproduce**
Steps to reproduce the behavior:
1. cd Firmware
2. make px4_sitl gazebo

dagar, 2020-06-03:
Could you try current master (soon to be v1.11)? If you still have a problem please share the ulg (https://logs.px4.io/).

deepak-1530, 2020-06-06:
Resolved the issue. It was happening because I had a separate build of sitl_gazebo and wasn't using the one that comes with the firmware when I updated it to 1.10.1. Just removed the earlier build of sitl_gazebo and set the environment variable for the new one.

Thanks

closed 2020-06-06 by deepak-1530