PX4/PX4-Autopilot #15028: Mac Toolchain Installation SHA256 Mismatch & Environment Issues

robinasu, 2020-06-04:
Hello, re-installing the dev toolchain from [this page](https://dev.px4.io/v1.9.0/en/setup/dev_env_mac.html) on MacOS 10.15.5 on 6/2/2020 resulted in some errors, similar to closed issue #14744. Are there any quick work-arounds?

Command `brew install px4-dev` led to the following issue :

`==> Downloading https://raw.githubusercontent.com/PX4/Firmware/master/Tools/px4. Already downloaded: /Users/robinsimpson 1/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py Error: SHA256 mismatch Expected: dafe6a32d270c48cbd42960adb2155bd5f60d1aec8cb644b10f8cafc0c8d2e8f Actual: 48199ee9ff392eff8a1efcd177e2c10f4a4dfd9877e52e13b1f3540d5dfedac9 Archive: /Users/robinsimpson 1/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py To retry an incomplete download, remove the file above.`

Command `brew install px4-sim` results in:

`==> Downloading https://raw.githubusercontent.com/PX4/Firmware/master/Tools/px4. Already downloaded: /Users/robinsimpson 1/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py Error: SHA256 mismatch Expected: dafe6a32d270c48cbd42960adb2155bd5f60d1aec8cb644b10f8cafc0c8d2e8f Actual: 48199ee9ff392eff8a1efcd177e2c10f4a4dfd9877e52e13b1f3540d5dfedac9 Archive: /Users/robinsimpson 1/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py To retry an incomplete download, remove the file above.`

Command `sudo -H pip install pyserial empy toml numpy pandas jinja2 pyyaml` resulted in:

`Using cached MarkupSafe-1.1.1-cp27-cp27m-macosx_10_6_intel.whl (17 kB)
Requirement already satisfied: six>=1.5 in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from python-dateutil>=2.5.0->pandas) (1.12.0)
ERROR: matplotlib 1.3.1 requires nose, which is not installed.
ERROR: matplotlib 1.3.1 requires tornado, which is not installed.
ERROR: pandas 0.24.2 has requirement numpy>=1.12.0, but you'll have numpy 1.8.0rc1 which is incompatible.
Installing collected packages: python-dateutil, pandas, MarkupSafe, jinja2, pyyaml
  Attempting uninstall: python-dateutil
    Found existing installation: python-dateutil 1.5
    Uninstalling python-dateutil-1.5:
ERROR: Could not install packages due to an EnvironmentError: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/parser.pyc', '/private/tmp/pip-uninstall-ZbKQHc/parser.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/parser.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/relativedelta.py', '/private/tmp/pip-uninstall-ZbKQHc/relativedelta.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/relativedelta.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/zoneinfo/zoneinfo-2010g.tar.gz', '/private/tmp/pip-uninstall-ZbKQHc/zoneinfo/zoneinfo-2010g.tar.gz', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/zoneinfo/zoneinfo-2010g.tar.gz'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/zoneinfo/__init__.py', '/private/tmp/pip-uninstall-ZbKQHc/zoneinfo/__init__.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/zoneinfo/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/zoneinfo/__init__.pyc', '/private/tmp/pip-uninstall-ZbKQHc/zoneinfo/__init__.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/zoneinfo/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/zoneinfo', '/private/tmp/pip-uninstall-ZbKQHc/zoneinfo', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/zoneinfo'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/tz.py', '/private/tmp/pip-uninstall-ZbKQHc/tz.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/tz.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/relativedelta.pyc', '/private/tmp/pip-uninstall-ZbKQHc/relativedelta.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/relativedelta.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/rrule.pyc', '/private/tmp/pip-uninstall-ZbKQHc/rrule.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/rrule.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/__init__.py', '/private/tmp/pip-uninstall-ZbKQHc/__init__.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/parser.py', '/private/tmp/pip-uninstall-ZbKQHc/parser.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/parser.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/tzwin.py', '/private/tmp/pip-uninstall-ZbKQHc/tzwin.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/tzwin.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/rrule.py', '/private/tmp/pip-uninstall-ZbKQHc/rrule.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/rrule.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/__init__.pyc', '/private/tmp/pip-uninstall-ZbKQHc/__init__.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/easter.py', '/private/tmp/pip-uninstall-ZbKQHc/easter.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/easter.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/easter.pyc', '/private/tmp/pip-uninstall-ZbKQHc/easter.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/easter.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/tz.pyc', '/private/tmp/pip-uninstall-ZbKQHc/tz.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/tz.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/tzwin.pyc', '/private/tmp/pip-uninstall-ZbKQHc/tzwin.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc/tzwin.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil', '/private/tmp/pip-uninstall-ZbKQHc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-ZbKQHc'")]`

julianoes, 2020-06-04:
Can you try to do:
```
brew update
brew upgrade
brew doctor
brew install px4-dev
```

And for the Python problems, check the newer docs: https://dev.px4.io/master/en/setup/dev_env_mac.html (not the 1.9 that you linked).

We now uses `pip3 install --user` in order to avoid permission problems like you're seeing.

jank324, 2020-06-04:
As a quick and dirty work-around you can just `brew edit` the both formulas locally and replace the actual hash value with the expected one from the error you are getting.

I have just tried to reproduce your issue and on my machine (macOS 10.15.5) it works just fine. Interestingly, your hashes are exactly the other way around compared to the issue I've had a while back (#14822) which was also related to the issue you mentioned.

I would suggest removing the already downloaded version of the files (as suggested in the error message) and then hopefully the hashes match once it downloads again. If this works, it would be a more proper and clean fix.

robinasu, 2020-06-04:
Ok, thank you for the confirmation re: 10.15.5. A couple of issues came up- First, I was not using the "master" documentation selector at the top left and was using the "stable" docs. Second, somehow my computer name now has a space and 1 at the end, and I could not delete the hash because of it; though it was an install issue. Annoying. I was able to reinstall everything, seemingly successfully, but Apple is complaining now:

BUILD SUCCESSFUL
Total time: 2 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

2020-06-04 11:04:07.794 java[15402:64888] Apple AWT Internal Exception: NSWindow drag regions should only be invalidated on the Main Thread!
2020-06-04 11:04:07.794 java[15402:64888] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff2f777be7 __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x00007fff6854f5bf objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff2f7a03dc -[NSException raise] + 9
	3   AppKit                              0x00007fff2c99eddc -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 310
	4   AppKit                              0x00007fff2c986842 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1416
	5   AppKit                              0x00007fff2c9862b3 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	6   libnativewindow_macosx.jnilib       0x0000000166798f6e Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
	7   ???                                 0x0000000108a91407 0x0 + 4440265735
)
libc++abi.dylib: terminating with uncaught exception of type NSException
/Users/robinsimpson 1/Firmware/Tools/jmavsim_run.sh: line 80: 15402 Abort trap: 6           java -XX:GCTimeRatio=20 -Djava.ext.dirs= -Djavax.accessibility.assistive_technologies=" " -jar jmavsim_run.jar $device $extra_args
1

I tried to reboot, no change. Have you seen this before?

robinasu, 2020-06-04:
Hey guys, I removed all my java versions and reinstalled rebooted, then verified java version:

`openjdk version "1.8.0_252"
OpenJDK Runtime Environment (AdoptOpenJDK)(build 1.8.0_252-b09)
OpenJDK 64-Bit Server VM (AdoptOpenJDK)(build 25.252-b09, mixed mode)`

I also reinstalled all the software in the "master" procedure after the open java install and still get the error in my previous comment.

I then swapped openjdk for Oraclejdk8 and reinstalled everything except openjdk and noticed another full Firmware folder gets copied into the proper Firmware folder- deleted both and re-cloned.. the jmavsim GUI now starts and takes commands. The first time the GUI loaded after I installed OracleJDK it would not take commands, so I started clean. Now the GUI takes commands and seems fine, but still complains in Terminal:

`Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily2020-06-04 14:08:31.496 java[16006:75656] WARNING: NSWindow drag regions should only be invalidated on the Main Thread! This will throw an exception in the future. Called from (
	0   AppKit                              0x00007fff2e5b1e19 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 371
	1   AppKit                              0x00007fff2e599842 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1416
	2   AppKit                              0x00007fff2e5992b3 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	3   libnativewindow_macosx.jnilib       0x0000000165cd3f6e Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
	4   ???                                 0x000000010c64c667 0x0 + 4502898279
)
INFO  [simulator] Simulator connected on TCP port 4560.`

Hope this helps somebody. :D

julianoes, 2020-06-05:
Thanks for the feedback.

Regarding jMAVSim:
https://dev.px4.io/master/en/simulation/jmavsim.html#troubleshooting
Feel free to add what you found there.

And if you want you can also test this pull request:
https://github.com/PX4/jMAVSim/pull/119