PX4/PX4-Autopilot #15031: catkin build fails for mavlink_sitl_gazebo

minutiaes, 2020-06-04:
After the command `make px4_sitl_default gazebo` in "Firmware" folder
I enter follow those,
`cd ~/ros_ws/src`
`ln -s Firmware/Tools/sitl_gazebo/`
`cd ~/ros_ws`
`catkin build`

while catkin builds everything, I got and strange issue and building fails.
`/home/berk/catkin_ws/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::VisionCallback(gazebo::OdomPtr&)’:
/home/berk/catkin_ws/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:1042:21: error: ‘MAV_FRAME_VISION_NED’ was not declared in this scope
     odom.frame_id = MAV_FRAME_VISION_NED;
`

I use ROS melodic with Gazebo version 9.0.0 under Ubuntu 18.04.

Thanks!

julianoes, 2020-06-04:
Can you try to update `sitl_gazebo` to latest master?
What version of PX4 are you using?

TSC21, 2020-06-04:
@minutiaes `catkin` builds fetch the ROS Mavlink package, which at this point probably is outdated. Though, what I can see above is that you are not using the latest `sitl_gazebo` master.

minutiaes, 2020-06-04:
> Can you try to update `sitl_gazebo` to latest master?
> What version of PX4 are you using?

this is the command how i clone it.
`git clone --branch stable https://github.com/PX4/Firmware.git`

TSC21, 2020-06-04:
> > Can you try to update `sitl_gazebo` to latest master?
> > What version of PX4 are you using?
> 
> this is the command how i clone it.
> `git clone --branch stable https://github.com/PX4/Firmware.git`

Yes that matches what I said - `MAV_FRAME_VISION_NED` was deprecated in https://github.com/mavlink/mavlink/pull/1112. Current Mavlink headers in ROS are synced with this deprecation, which means that you rather downgrade the Mavlink ROS package headers to a compatible version with the stable release, or you use the `master` branch of the Firmware.

closed 2020-09-02 by TSC21