PX4/PX4-Autopilot #15037: tailsitter vtol forward transmission failed (no pitch set value is generated)

xdwgood, 2020-06-05:
**Describe the bug**
When I used tailsitter to fly, I found that **sometimes**(Sometimes it can enter successfully, in the same flight log) even if I switch, vtoll can not enter the fw mode

**To Reproduce**
1. use master branch fly(tailsitter vtol)
2. in altitude mode and try transmission(mc to fw) 
3. The drone pitch angle has not changed, there is no tilt acceleration

**Expected behavior**
Every time a customer uses a stick for transmission, vtol should enter the transmission state correctly

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=4df87c34-da2f-46be-b612-ddcc13b135c8

[image]

**Additional context**
I found that fw to mc sometimes fails when I try to send a transfer command using the remote control. I will do some investigations to see where the problem is. I now suspect that the transmission command is not correctly accepted in tailsitter.cpp (may be uorb related)

FYI: @sfuhrer  :smiley:

xdwgood, 2020-06-05:
Is there any problem with the numerical calculations under certain extreme conditions? (At the start of transmission, the actual angle of pitch and the set angle are both 0)

here:   https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/tailsitter.cpp#L206-L218

Because I compared the two switches, the only difference I found was that the drone angle/setpoint angle was close to 0 before the failed transmission started

xdwgood, 2020-06-05:
When the transmission is run in altitude mode, the setpoint angle value generated by mc and fw is not published　herer: https://github.com/PX4/Firmware/blob/master/src/modules/mc_att_control/mc_att_control_main.cpp#L329
https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L141

So I am curious about the meaning of the check here: https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L436

On the other hand about this check: `updated() `should be used, right?(Instead of` update()`)　https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L401-L402

My idea is to remove this check because the angle setting in the conversion stage is generated independently by vtol control....

Back to this question, I suspect that this has not been run for some reason:　https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L437

sfuhrer, 2020-06-09:
@xdwgood Did I understand correct, that the issue is this spike in the transition switch log?
[image]
Or what exactly? If it's the switch, then this is not really VTOL related, but rather a RC problem no?

xdwgood, 2020-07-16:
@sfuhrer  
https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/tailsitter.cpp#L216

The calculated `tilt` sometimes has problems. I checked it using the `PX4_ISFINITE(tilt)` function and found this problem.
[image]

Corresponding code

[image]

xdwgood, 2020-07-16:
No setpoint value is generated when the value check fails(8)

xdwgood, 2020-07-17:
fix here: https://github.com/PX4/Firmware/pull/15352!

closed 2020-07-17 by xdwgood