PX4/PX4-Autopilot #15042: Master: scaled IMU high noise

almaaro, 2020-06-05:
**Describe the bug**
MAV topic SCALED IMU shows very high noise. Amplitude is about 1000. Unable to calibrate gyro (movement detected)

**To Reproduce**
Steps to reproduce the behavior:
- Master 5.6.2020
- Pixracer R14
- Standard plane

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
[image]

https://logs.px4.io/plot_app?log=dfc4f4f4-ce6b-4f57-bece-88d0a4f229f7

dagar, 2020-06-05:
There are a lot of errors on the first IMU (icm20608g).

[image]

This might already be fixed in https://github.com/PX4/Firmware/pull/15004. Could you please give it a quick try?

almaaro, 2020-06-05:
Tested #154004, but the problem still persists.

dagar, 2020-06-05:
> Tested #15004, but the problem still persists.

Thanks for the quick test. Could you share a log again with #15004 or simply grab some console output?

``` Console
icm20608g status

perf

listener sensor_accel

listener sensor_accel_fifo

sensors status
```

I have a pixracer with an icm20608g, but I haven't been able to reproduce this failure.

almaaro, 2020-06-05:
As requested. 
https://logs.px4.io/plot_app?log=6b1737af-4b6c-4cc0-96ff-0b730dbfd946

dagar, 2020-06-05:
> As requested.
> https://logs.px4.io/plot_app?log=6b1737af-4b6c-4cc0-96ff-0b730dbfd946

Very odd. Within #15004 could you try cutting the bus frequency in half? 
https://github.com/PX4/Firmware/blob/6063fce6738884d4681d05c5e59cdc4d387a194c/src/drivers/imu/invensense/icm20608g/InvenSense_ICM20608G_registers.hpp#L57

If it's still problematic after that try running without the mpu9250 by either stopping it manually or commenting it out here. https://github.com/PX4/Firmware/blob/6063fce6738884d4681d05c5e59cdc4d387a194c/boards/px4/fmu-v4/default.cmake#L35

These Invensense IMU drivers (new since v1.10) are doing significantly more work transferring all 8 kHz raw data. What's a bit odd here is the ICM20602 (on most pixracers and very similar to the ICM20608g) has received significant testing across many different flight controllers and doesn't have any of these problems.

almaaro, 2020-06-05:
Cutting the bus frequency and disabling the MPU9250 didn't work. Still similar output, FIFO overflows, transfer errors and sensor status is TOUT ECNT.

Could it be a hardware issue? I'm using a clone pixracer from banggood.

dagar, 2020-06-05:
> Could it be a hardware issue? I'm using a clone pixracer from banggood.

Likely, but I'd like to see if we can keep it working.

I'm wondering if a much longer reset will help. Increase this to 500?
https://github.com/PX4/Firmware/blob/b607630ee751e0817a46998dc862f5e5ab69f7f8/boards/px4/fmu-v4/init/rc.board_sensors#L8-L14

dagar, 2020-06-05:
It looks like it's a problem for both IMUs (0: icm20608g and 1: mpu9250).

``` Console
nsh> listener sensor_accel

TOPIC: sensor_accel 2 instances

Instance 0:
 sensor_accel_s
	timestamp: 150645406  (0.013041 seconds ago)
	timestamp_sample: 150644751  (655 us before timestamp)
	device_id: 3801098 (Type: 0x3A, SPI:1 (0x00)) 
	x: -11.1677
	y: -15.5207
	z: 23.3441
	temperature: 0.0000
	error_count: 87396
	clip_counter: [0, 0, 0]

Instance 1:
 sensor_accel_s
	timestamp: 150659327  (0.009868 seconds ago)
	timestamp_sample: 150658978  (349 us before timestamp)
	device_id: 2359306 (Type: 0x24, SPI:1 (0x00)) 
	x: -2.1652
	y: 0.3220
	z: -22.9504
	temperature: 52.8208
	error_count: 2
	clip_counter: [0, 0, 0]
```

dagar, 2020-06-05:
I think I've got it. For some reason on certain (older?) ICM20608g units you can't read the FIFO count in the same long transfer as the actual FIFO. This is consistent with older models like the mpu6000.

Could you try the current https://github.com/PX4/Firmware/pull/15004 one more time? https://github.com/PX4/Firmware/pull/15004

almaaro, 2020-06-05:
You've surely got it! No errors anymore. Thanks!

closed 2020-06-05 by almaaro