PX4/PX4-Autopilot #15043: OneShot broken on master

bkueng, 2020-06-05:
This is how the signal looks like, and apparently the ESC's don't like it:
[image]
The problem is the 2 callbacks in the pwm driver, the rate controller output and RC (the signal is fine when shutting off RC).

I have not checked when it broke exactly, just that it is not aa380f351fa01edfce71533c7eaefec4ff6446ee.

dagar, 2020-06-05:
Two immediate ideas that come to mind.

1. Add a special case in the `mixer_module` to not register additional callbacks after actuator_controls_0/actuator_controls_1 if they're part of the mix.
2. Add a low level (minimum interval?) protection for `oneshot`. Are there well defined requirements?

dagar, 2020-06-06:
I'll also add a simple minimum interval (maximum rate) mechanism to `WorkItem`.

commit 60fe2077 (2020-07-20): pwm_out: limit actuator control callbacks to primary groups

 - add backup schedule at minimum PWM rate
 - fixes #15043
commit aa7735c1 (2020-07-23): pwm_out: limit actuator control callbacks to primary groups

 - add backup schedule at minimum PWM rate
 - fixes #15043
closed 2020-07-23 by dagar