PX4/PX4-Autopilot #15045: lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.29' not found

0b10010010, 2020-06-05:
Hello,

I am building emlid_navio2_cross using ubuntu 20.04 and when I upload to RPi and start px4 using `sudo ./emlid_navio2_cross/bin/px4` I get an error saying ```./emlid_navio2_cross/bin/px4: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.29' not found (required by ./emlid_navio2_cross/bin/px4)```

From what I found my RPi has libc 2.28 installed which is the most current version.

Any help is much appreciated.

dagar, 2020-06-07:
I'd recommend you either build within a docker container with the same version of glibc or compile directly on the raspberry pi.

Install Docker and then from the top level of Firmware you can run this shortcut. https://hub.docker.com/r/px4io/px4-dev-armhf

``` Console
$ ./Tools/docker_run.sh `make emlid_navio2_default`
```
Note, in newer PX4 `emlid_navio2_cross` and `emlid_navio2_native` have merged into `emlid_navio2_default`.

0b10010010, 2020-06-08:
Thanks for your reply @dagar. I just went back to using previous environment with Mint 19.3 and everything works fine now. Thanks again for your help.

closed 2020-06-08 by 0b10010010

karimahmadidastgerdi, 2021-07-11:
Hi dear,
I have this problem too
could you please explain more how to build an upload px4 for navio2 within docker
Thanks

dagar, 2021-07-12:
Did you try installing docker and using the command mentioned above?

https://github.com/PX4/PX4-Autopilot/issues/15045#issuecomment-640233263

karimahmadidastgerdi, 2021-07-12:
Thanks for replying/
yes I did it and it build px4/
now I do not know how to upload it on the board/