PX4/PX4-Autopilot #15049: File access via usb doesn't´t work anymore

taileron, 2020-06-07:
(due a crash I had to glue in the sd card into the PixRacer, so I have to be able to change files via usb)
if the sd card was plugged into a Mac one time, folders and files like .Spotlight-V100 get created with filenames longer than 32 characters. This disturbs all file exchange between computer and FC and they cannot be deleted easily.
all standard attempts went wrong
/fs/microsd:
 dataman
 log/
 params
 .Spotlight-V100/
 logdata.txt
 etc/
 mission_log/

nsh> rmdir .Spotlight-V100
nsh: rmdir: rmdir failed: Directory not empty
nsh> cd .Spotlight-V100
nsh> ls
/fs/microsd/.Spotlight-V100:
 Store-V2/
nsh> cd Store-V2
nsh> ls
/fs/microsd/.Spotlight-V100/Store-V2:
 2BA7B381-D2CC-4E42-87C9-1A85A2B7/
nsh> cd 2BA7B381-D2CC-4E42-87C9-1A85A2B7
nsh: cd: chdir failed: No such file or directory
nsh> cd "2BA7B381-D2CC-4E42-87C9-1A85A2B7"
nsh: cd: chdir failed: No such file or directory
nsh> rmdir 2BA7B381-D2CC-4E42-87C9-1A85A2B7
nsh: rmdir: rmdir failed: No such file or directory
nsh> rmdir "2BA7B381-D2CC-4E42-87C9-1A85A2B7"
nsh: rmdir: rmdir failed: No such file or directory

taileron, 2020-06-07:
To get rid of this folders that don´t have to be empty simply rename them into XXX or ZZZ with capital letters and then rename them to the same name with lower case letters that deletes them inclusive all of their content.
Open mavlink console and type:  
nsh> mv .Spotlight-V100 ZZZ
nsh> mv ZZZ zzz 

That deletes the disturbing folders and files without refusal because not empty or too long filenames.
Afterwards the file exchange e.g. with files widget works perfectly again.

dagar, 2020-06-08:
>  This disturbs all file exchange between computer and FC and they cannot be deleted easily.

Could you expand on this? Are you talking about this being problematic for Mavlink FTP or only when using the command line (nsh)? Was this working in an older PX4 release?

taileron, 2020-06-08:
This Apple files and folders disturb different ways of file communication via usb on current and old versions of PX4. (nsh) and e.g. files widget (QGC up to 3.5x, 4.xx does´t offer that function anymore). It concerns only this Apple files that do not belong to PX4 and should not be on the SD.
But if this SD was plugged on time into an Mac then they are there, block the file transfer and cannot be deleted, they could only be deleted when plugged into a Windows machine with the file system is set to all files visible ....  this requires the SD can be removed out of the FC. 
I spent a whole day to figure that out .... cause my sd card is glued into the Pixracer.
The (nsh) mv capital to small letters command deletes folders even there are too long filenames included that cannot be deleted anymore with (nsh) rm or rmdir. This solution for the simple deletion of unwanted folders on the SD could well end up in the documentation to help others too.

dagar, 2020-06-08:
Can you try https://github.com/PX4/Firmware/pull/15051?

taileron, 2020-06-09:
As soon as there are real index folders that content files with longer filenames than the PX4 supported 32 letters it  still disturbs the file management.
[image]
That makes it more complicated as you see now the original folders and the PX generated ones.

dagar, 2020-06-09:
We could add something in mavlink FTP. I'm wondering if we could get away with ignoring all of these hidden (.*) files. 

https://github.com/PX4/Firmware/blob/13d983d30506c5d4ad66888438246b00ca05ec97/src/modules/mavlink/mavlink_ftp.cpp#L415-L433

julianoes, 2020-06-10:
I think we should not do the filtering on the MAVLink FTP server side but the client can filter them out if they are not interested in hidden files.

taileron, 2020-06-11:
probably I used the wrong platform to show problem and solution ... as soon as .Spotlight-V100 is removed, it always has contents with too long file names, there are no more problems at all.
I think there is no way to prevent it on the PX4 side, cause (nsh) will now not suddenly be able to manage longer file names.
It should remain as it was before, only the documentation could point out that existing folders on the SD card with contents that the PX4 file system cannot handle may cause problems and should be deleted first.

dagar, 2020-06-11:
> I think there is no way to prevent it on the PX4 side, cause it will now not suddenly be able to manage longer file names.

We can't outright prevent it, but won't this eliminate the problem in most cases going forward? Boards typically start with a fresh SD card and PX4 will create the files that block Spotlight right away.

taileron, 2020-06-11:
the Spotlight folder can hide content in subfolders that do not follow the PX4 file system conventions. 
That happens only after you had plugged the sd card physically into a Mac.
A fresh Sd card can not be affected by this, and PX4 doesn't´t create any unwanted files at all. 
PX4 only cannot handle some folders that came onto the SD card when it was outside the PX4 device.

taileron, 2021-01-19:
In various ways one comes across the old problem again ... the lines 113 - 137 in RCS should be removed again, as they do not help except to disguise real problems. #16334

closed 2021-11-25 by taileron