PX4/PX4-Autopilot #15054: SITL CI VTOL tailsitter intermittent failures

dagar, 2020-06-08:
- http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests_ASan/detail/master/1851/pipeline

[image]

Is this a VTOL problem or an issue with the simulation?

dagar, 2020-06-08:
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests_coverage/detail/pr-sens_ext_i2c_probe/1/pipeline

Jaeyoung-Lim, 2020-06-08:
@dagar Those resets of thrust looks strnage. Is there something publishing zero thrust?

dagar, 2020-06-08:
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests_coverage/detail/pr-boards_cleanup/4/pipeline

sfuhrer, 2021-01-04:
I don't think this is still valid, is it?

Jaeyoung-Lim, 2021-04-05:
I believe this is not valid anymore, closing

closed 2021-04-05 by Jaeyoung-Lim