PX4/PX4-Autopilot #15065: GPS lost in position control lands without pilot having control

julianoes, 2020-06-09:
**Describe the bug**
When GPS is lost while flying in POSCTL I expect the vehicle to go into ALTCTL so that the pilot can still bring the vehicle safely back. However, what happens instead is that a blind land is initiated, even though the mode displayed in QGC is ALTCTL.

**To Reproduce**
Steps to reproduce the behavior:
1. Use: https://github.com/PX4/Firmware/pull/15063
2. Takeoff and fly in POSCTL (e.g. using Joystick in QGC).
3. Enable failure injection: `param set SYS_FAILURE_EN 1`
4. Trigger GPS lost: `failure gps off`

**Log Files and Screenshots**

https://logs.px4.io/plot_app?log=8bd577fe-63ea-41da-8846-33f2fef20db5

```
pxh> param set SYS_FAILURE_EN 1
  SYS_FAILURE_EN: curr: 0 -> new: 1
pxh> 
pxh> 
pxh> failure gps off
pxh> WARN  [commander] Failsafe enabled: no local position
INFO  [commander] Failsafe mode activated
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] Failsafe: blind land
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] Failsafe: blind land
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] Failsafe: blind land
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] Failsafe: blind land
WARN  [ecl/EKF] 156392000: GPS data stopped
INFO  [ecl/EKF] 156392000: reset position to last known position
INFO  [ecl/EKF] 156392000: reset velocity to zero
WARN  [ecl/EKF] 156392000: stopping navigation
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] Failsafe: blind land
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
INFO  [commander] Landing detected
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] Failsafe: blind land
INFO  [commander] Takeoff detected
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] PositionControl: invalid setpoints
INFO  [commander] Landing detected
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] Failsafe: blind land
WARN  [mc_pos_control] PositionControl: invalid setpoints
INFO  [commander] DISARMED by Auto disarm initiated
INFO  [commander] Failsafe mode deactivated
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN  [mc_pos_control] Previous flight task failed, switching to mode 1
INFO  [logger] closed logfile, bytes written: 9569092
```

FYI @MaEtUgR @dagar

MaEtUgR, 2020-06-09:
This quite bad :scream: It looks like whatever `Failsafe mode activated` is setting horizontal position/velocity setpoints for which there is no estimate. The repeating error messages are a last resort failsafe that makes sure the drone stays under control and should not happen.

MaEtUgR, 2020-06-09:
I found the main symptom in the setpoints. It's not setpoints with no estimate but the failsafe is Altitude mode and for some reason it's not publishing any horizontal setpoints:
[image]
That's why the error handler has to take over and emergency land the vehicle.

closed 2020-06-09 by dagar

MaEtUgR, 2020-06-09:
Altitude mode is broken on master **IF** `activate()` and the first `update()` of the task happen exactly at the same time which is likely only possible in SITL where two `hrt_absolute_time()` calls ran tens of lines of code after each other always have the exact same value and always the last three digits zero e.g.: `4564000 4564000`. I'm guessing it "only" measures miliseconds. In any case the problem of the delta time being exactly zero needs to be generally prevented and that's why I made this pr: https://github.com/PX4/ecl/pull/842

Can we get that in quickly?

MaEtUgR, 2020-06-10:
I double-checked this morning on master and the issue is not reproducible for me. Thanks @julianoes and @dagar!

julianoes, 2020-06-10:
Oh wow, thanks @MaEtUgR!