PX4/PX4-Autopilot #15066: GPS lost during VTOL fixedwing mission immediately triggers loiter descent

julianoes, 2020-06-09:
**Describe the bug**
When GPS is lost during the fixedwing part of a VTOL mission, the plane says it is going to `fixed bank loiter`. It does so without thrust, so starts diving to keep above stall speed. This results in a "hard landing" at 13 m/s horizontal and 3 m/s vertical speed.

I would assume that by default a VTOL would either actually loiter for a while to try to regain GPS or otherwise transition back to multicopter mode and slowly descend.

It certainly should not dive like that, especially when datalink and manual control is still available.

**To Reproduce**
Steps to reproduce the behavior:
1. Use: https://github.com/PX4/Firmware/pull/15063
2. `make px4_sitl gazebo_standard_vtol`
3. Plan VTOL mission in QGC and start it.
4. Enable failure injection: `param set SYS_FAILURE_EN 1`
5. Trigger GPS lost: `failure gps off`

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=d4571b45-9472-4265-bc6c-87555e95c784

```
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
  MAV_TYPE: curr: 2 -> new: 22
INFO  [init] Mixer: etc/mixers-sitl/standard_vtol_sitl.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2020-06-09/15_20_07.ulg
INFO  [logger] Opened full log file: ./log/2020-06-09/15_20_07.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
INFO  [ecl/EKF] 568000: GPS checks passed
INFO  [ekf2] Mag sensor ID changed to 197388
INFO  [ecl/EKF] 1792000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 1792000: reset position to last known position
INFO  [ecl/EKF] 1792000: reset velocity to zero
INFO  [ecl/EKF] 5151999: reset position to GPS
INFO  [ecl/EKF] 5151999: reset velocity to GPS
INFO  [ecl/EKF] 5151999: starting GPS fusion
INFO  [commander] ARMED by Arm/Disarm component command
INFO  [navigator] Takeoff to 50.0 meters above home.
INFO  [commander] Takeoff detected

pxh> 
pxh> 
pxh> 
pxh> failure gps off
ERROR [failure] Failure injection disabled by SYS_FAILURE_EN param.
Command 'failure' failed, returned 1.
pxh> param set SYS_FAILURE_EN 1
  SYS_FAILURE_EN: curr: 0 -> new: 1
pxh> failure gps off
pxh> 
pxh> 
pxh> WARN  [commander] Failsafe enabled: no global position
INFO  [commander] Failsafe mode activated
WARN  [navigator] Global position failure: fixed bank loiter
WARN  [ecl/EKF] 156320000: GPS data stopped
INFO  [ecl/EKF] 156320000: reset position to last known position
INFO  [ecl/EKF] 156320000: reset velocity to zero
WARN  [ecl/EKF] 156320000: stopping navigation

pxh> commander disarm
pxh> WARN  [commander] Disarming denied! Not landed

pxh> 
pxh> commander disarm -f
pxh> WARN  [commander] Disarming denied! Not landed
```

ryanjAA, 2020-10-07:
This actually just happened to me and I wasn’t prepared for the default behavior. My question would be why by default is NAV_GPSF_TR 0? The answer I’m received was to descend but is that correct option as well as default setting (when say rc and or telemetry is available?) Thad depends also on NAV_GPSF_LT (which prob shouldn’t default to 0) but why thrust is 0 doesn’t make sense. Am I missing something here or is the objective to hard land aka fixed bank loiter to land with no throttle?

bernhardpg, 2021-02-02:
We experienced this on our BabyShark 260 VTOL a few days back. GPS position was lost, triggering the `condition_local_velocity_valid`, `condition_local_position_valid` and `condition_global_position_valid` flags, which in turn triggered the nav_state to be set to `NAVIGATION_STATE_AUTO_LANDGPSFAIL`, which gives the "fixed bank loiter" described here. As the vehicle was descending quite quickly, we did a transition from FW to RW mode manually in QGC. This immediately triggered the vehicle to change the nav_state to `NAVIGATION_STATE_DESCEND`, causing it to land (thankfully safely) in RW mode.

The log can be found [here](https://logs.px4.io/plot_app?log=9add6c94-ce95-4992-8c66-dbc96f48e239). This was not at all the expected behaviour, as we were assuming that the vehicle would either loiter for a while or transition to RW mode, as @julianoes describes.

@julianoes @sfuhrer What is the status on this issue? Is a fix for this behaviour planned in the near future?

bernhardpg, 2021-02-02:
Also, it seems that the altitude is set to 0 when this happens, as `reset_triplets()` is called by the navigator in `navigator_main.cpp` when the nav_state is changed to fixed bank loiter (`NAVIGATION_STATE_AUTO_LANDGPSFAIL`). Could it be that this is what is causing the motor to give zero thrust @julianoes ?

[image]

julianoes, 2026-08-31:
Might be worth checking still...

dakejahl, 2026-11-01:
**Claude on behalf of @dakejahl**

Fixed on main by #17547: losing GPS now switches to Descend. A VTOL in fixed-wing flight transitions to hover (NAV_FORCE_VT, on by default), and a fixed-wing loiters at altitude for FW_GPSF_LT before descending.

closed 2026-11-01 by dakejahl

julianoes, 2026-11-01:
@dakejahl this bot is so much better than the stupid stale bot.