PX4/PX4-Autopilot #15088: SITL: Logger is missing samples of ekf2_timestamps

RomanBapst, 2020-06-11:
When running SITL and adding a counter to the ekf2_timestamps topic we realized that not all the samples are logged. The logger does not report and dropouts though.
I went back to v1.9.2 and still saw the same behavior.
We are thinking that this might be the reason why replay on SITL logs is not working well.

[image]

julianoes, 2020-06-11:
I think the problem is that lockstep only waits for data ready to be sent to the simulator, so the actuator_controls essentially. However, it doesn't wait until logger is done logging everything.

We could think of adding some sort of flag for logger to signal that it is done writing things to disk, so that we can continue with lockstep.

bkueng, 2020-06-11:
Yes we need to integrate the logger into the lockstep cycle to solve this cleanly.

However a quick test showed that also letting logger poll on `ekf2_timestamps` improves things a lot already.

closed 2020-06-23 by dagar