PX4/PX4-Autopilot #15090: After the X7 flight control board is configured with 14 PWM channels, an error (this statement may fall through) will be prompted when compiling the firmware

CUAVcaijie, 2020-06-12:
**Describe the bug**
After the X7 flight control board is configured with 14 PWM channels, an error (this statement may fall through) will be prompted when compiling the firmware
[image]

**To Reproduce**
Test 14 PWM source addresses ：
https://github.com/cuav/Firmware/tree/pr-pwm14

I commented some code and compiled it. The compilation was successful. I want to know what caused this error.
[image]

URL after commenting code：
https://github.com/cuav/Firmware/tree/pr-pwm14_1

dagar, 2020-06-15:
As you noticed I left this for later in the initial x7pro board support. Let's open a PR and take a closer look. 

This also reminds me we didn't get Nora support merged. https://github.com/PX4/Firmware/pull/14475
We should bring over the recent x7pro changes/fixes and get it in.

dagar, 2020-06-15:
Continued in https://github.com/PX4/Firmware/pull/15110.

cuhome, 2020-06-19:
@dagar I tested the pr-pwm14_1 branch firmware, Generic Quadplane VTOL frame aux1~aux5 have no signal output.

jinger26, 2021-01-14:
@dagar  any update on this?

closed 2021-03-26 by CUAVcaijie