PX4/PX4-Autopilot #15091: error while " make px4_sitl gazebo" on macOS

kainism, 2020-06-12:
OS: macOS catalina 10.15.5
Firmware: updated

kevin@kevinH'sMacBook-Pro ~ % cd firmware
kevin@kevinH'sMacBook-Pro firmware % make px4_sitl gazebo
sed: 1: "s/.*48885.*/Application ...": bad flag in substitute command: 'X'
[0/4] Performing build step for 'sitl_gazebo'
[1/72] Building CXX object CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o
FAILED: CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBBULLET_VERSION=2.89 -DLIBBULLET_VERSION_GT_282 -Dnav_msgs_EXPORTS -I/Users/kevin/Firmware/Tools/sitl_gazebo/include -I. -I/usr/local/include/eigen3/eigen3 -I/usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs -I/Users/kevin/Firmware/mavlink/include -I/Users/kevin/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/local/Cellar/libffi/3.3/include -I/usr/local/Cellar/gstreamer/1.16.2/include/gstreamer-1.0 -I/usr/local/Cellar/glib/2.64.3/include -I/usr/local/Cellar/glib/2.64.3/include/glib-2.0 -I/usr/local/Cellar/glib/2.64.3/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.44/include -isystem /usr/local/Cellar/gazebo9/9.13.0_1/include -isystem /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9 -isystem /usr/local/Cellar/bullet/2.89/include/bullet -isystem /usr/local/Cellar/bullet/2.89/include -isystem /usr/local/include/simbody -isystem /usr/local/include -isystem /usr/local/Cellar/sdformat5/5.3.0_5/include/sdformat-5.3 -isystem /usr/local/include/ignition/math3 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Paging -isystem /usr/local/include/ignition/math4 -isystem /usr/local/include/ignition/transport4 -isystem /usr/local/include/ignition/msgs1 -isystem /usr/local/include/ignition/common1 -isystem /usr/local/include/ignition/fuel_tools1 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/opencv/4.3.0/include/opencv4 -isystem /usr/local/Cellar/fcl/0.6.1/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/Cellar/ffmpeg/4.2.3_1/include -Wno-deprecated-declarations -Wno-address-of-packed-member -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -fPIC   -std=c++11 -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -I/usr/local/Cellar/ffmpeg/4.2.3_1/include -std=gnu++17 -MD -MT CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o -MF CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o.d -o CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o -c Odometry.pb.cc
In file included from Odometry.pb.cc:4:
In file included from ./Odometry.pb.h:35:
/usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs/quaternion.pb.h:52:51: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::PROTOBUF_NAMESPACE_ID::internal::AuxillaryParseTableField aux[]
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
In file included from Odometry.pb.cc:4:
In file included from ./Odometry.pb.h:36:
/usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs/vector3d.pb.h:52:51: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::PROTOBUF_NAMESPACE_ID::internal::AuxillaryParseTableField aux[]
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
2 errors generated.
[2/72] Building CXX object CMakeFiles/gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o
FAILED: CMakeFiles/gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBBULLET_VERSION=2.89 -DLIBBULLET_VERSION_GT_282 -Dgazebo_gimbal_controller_plugin_EXPORTS -I/Users/kevin/Firmware/Tools/sitl_gazebo/include -I. -I/usr/local/include/eigen3/eigen3 -I/usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs -I/Users/kevin/Firmware/mavlink/include -I/Users/kevin/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/local/Cellar/libffi/3.3/include -I/usr/local/Cellar/gstreamer/1.16.2/include/gstreamer-1.0 -I/usr/local/Cellar/glib/2.64.3/include -I/usr/local/Cellar/glib/2.64.3/include/glib-2.0 -I/usr/local/Cellar/glib/2.64.3/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.44/include -isystem /usr/local/Cellar/gazebo9/9.13.0_1/include -isystem /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9 -isystem /usr/local/Cellar/bullet/2.89/include/bullet -isystem /usr/local/Cellar/bullet/2.89/include -isystem /usr/local/include/simbody -isystem /usr/local/include -isystem /usr/local/Cellar/sdformat5/5.3.0_5/include/sdformat-5.3 -isystem /usr/local/include/ignition/math3 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Paging -isystem /usr/local/include/ignition/math4 -isystem /usr/local/include/ignition/transport4 -isystem /usr/local/include/ignition/msgs1 -isystem /usr/local/include/ignition/common1 -isystem /usr/local/include/ignition/fuel_tools1 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/opencv/4.3.0/include/opencv4 -isystem /usr/local/Cellar/fcl/0.6.1/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/Cellar/ffmpeg/4.2.3_1/include -Wno-deprecated-declarations -Wno-address-of-packed-member -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -fPIC   -std=c++11 -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -I/usr/local/Cellar/ffmpeg/4.2.3_1/include -std=gnu++17 -MD -MT CMakeFiles/gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o -MF CMakeFiles/gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o.d -o CMakeFiles/gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o -c /Users/kevin/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp
In file included from /Users/kevin/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:18:
In file included from /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/physics/physics.hh:2:
In file included from /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/physics/Actor.hh:26:
In file included from /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/physics/Model.hh:30:
In file included from /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/physics/Entity.hh:28:
In file included from /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs/msgs.hh:35:
In file included from /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs/MessageTypes.hh:7:
[... cut ...]
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/kevin/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.3/bin/cmake --build .
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

julianoes, 2020-06-24:
Have you tried `brew upgrade` and then `make clean` and `make px4_sitl gazebo` again?

jbeyerstedt, 2020-06-25:
This seems to be the related to https://github.com/PX4/sitl_gazebo/issues/513, which is related to https://github.com/osrf/homebrew-simulation/issues/1042. Protobuf renamed the `AuxillaryParseTableField` to `AuxiliaryParseTableField` field in version 3.12.3.

As stated in the sitl_gazebo issue, running `brew reinstall --build-from-source gazebo` helps until the underlying issue is resolved.

I have the same issue on my machine. Installed version of protobuf is 3.12.3. The only solution, that worked, was to rebuild gazebo from source. I tried to update the sitl_gazebo submodule to the latest master and to reinstall `px4-sim-gazebo` and `protobuf` beforehand, but had no success with that.

Jaeyoung-Lim, 2020-11-19:
This has been fixed, and the CI was added in https://github.com/PX4/PX4-SITL_gazebo/pull/613

closed 2020-11-19 by Jaeyoung-Lim