PX4/PX4-Autopilot #15093: PX4 v1.10 Mavros offboard landing confusion

nbelanger99, 2020-06-12:
We have a quad with quite good EV feedback from a mocap system, have `EKF2_AID_MASK` set to decimal 24, and by publishing to `setpoint_position/local` while in `OFFBOARD` mode, we have pretty good in-air position control over the drone. 

We've tried using the `LAND` mode, but found that it's too harsh and not precise enough. We're trying a velocity-based approach, publishing to `setpoint_velocity/cmd_vel_unstamped` while still in offboard mode to command the drone to move straight down. This will land the drone (and activate the landing detector), but it's not very precise. Something weird happens where it seems that PX4 starts to ignore yaw commands?
[image]

Should I be publishing to `setpoint_raw` rather than `setpoint_velocity/cmd_vel_unstamped`?
Should we be using mocap for velocity feedback as well?
Is there a way to do something like the precision landing mode using mocap position feedback? 

Seems like I'm missing something here; I feel like landing accurately with an EV system should be somewhat straightforward.

julianoes, 2020-06-24:
Please add a log to logs.px4.io and link it here.

nbelanger99, 2020-06-25:
Here's the flight log: https://review.px4.io/plot_app?log=c290d0d7-f02a-446d-b2c5-07a31582ed99

Also, I've opened up a mavros issue related to this: https://github.com/mavlink/mavros/issues/1440

echoGee, 2020-10-05:
Any updates on this ?