PX4/PX4-Autopilot #15102: unable to start the simulator either on gmavsim as well as gazebo with different errors

KumarSaurav366, 2020-06-14:
Saurav@Kumars-MacBook-Air-94 ~ % cd Firmware          
Saurav@Kumars-MacBook-Air-94 Firmware % make px4_sitl jmavsim
sed: 1: "s/.*2575.*/Applications ...": bad flag in substitute command: 'X'
[  1%] Built target lockstep_scheduler
[  2%] Built target logs_symlink
[  2%] Built target drivers_board
[  2%] Built target uorb_headers
[  2%] Built target git_ecl
[  2%] Built target ver_gen
[  2%] Built target git_mavlink_v2
[  2%] Built target mixer_gen_6dof
[  3%] Built target mixer_gen
[  3%] Built target git_gps_devices
[  4%] Built target rc
[  4%] Built target git_jmavsim
[  4%] Built target tinybson
[  5%] Built target perf
[  7%] Built target work_queue
[  7%] Built target ecl_airdata
[  8%] Built target ecl_geo_lookup
[  8%] Built target ecl_geo
[  8%] Built target ecl_tecs
[  9%] Built target ecl_validation
[  9%] Built target version
[  9%] Built target Mixer
[ 10%] Built target parameters
[ 29%] Built target uorb_msgs
[ 29%] Built target ecl_l1
[ 31%] Built target ecl_EKF
[ 31%] Built target SimpleMixer
[ 31%] Built target HelicopterMixer
[ 32%] Built target MultirotorMixer
[ 32%] Built target NullMixer
[ 33%] Built target generate_topic_listener
[ 34%] Built target modules__uORB
[ 34%] Built target mixer
[ 35%] Built target px4_work_queue
[ 36%] Built target px4_platform
[ 37%] Built target px4_daemon
[ 38%] Built target px4_layer
[ 39%] Built target systemlib
[ 40%] Built target cdev
[ 40%] Built target bezier
[ 40%] Built target battery
[ 40%] Built target lib__cdev__test__cdev_test
[ 40%] Built target circuit_breaker
[ 40%] Built target airspeed
[ 40%] Built target AirspeedValidator
[ 41%] Built target hysteresis
[ 43%] Built target CollisionPrevention
[ 44%] Built target conversion
[ 44%] Built target FlightTask
[ 44%] Built target lib__controllib__controllib_test
[ 45%] Built target controllib
[ 45%] Built target drivers__device
[ 45%] Built target drivers_magnetometer
[ 45%] Built target drivers_barometer
[ 46%] Built target Takeoff
[ 47%] Built target systemcmds__perf
[ 47%] Built target systemcmds__pwm
[ 47%] Built target systemcmds__sd_bench
[ 48%] Built target examples__hello
[ 49%] Built target FlightTaskOffboard
[ 50%] Built target FlightTaskFailsafe
[ 50%] Built target FlightTaskDescend
[ 50%] Built target FlightTaskTransition
[ 51%] Built target modules__landing_target_estimator
[ 52%] Built target modules__logger
[ 53%] Built target modules__local_position_estimator
[ 53%] Built target pid
[... 238 lines cut ...]

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
[ 41%] Built target circuit_breaker
See also "/Users/Saurav/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
[ 41%] Built target airspeed
make[4]: *** [external/Stamp/sitl_gazebo/sitl_gazebo-configure] Error 1
make[3]: *** [platforms/posix/CMakeFiles/sitl_gazebo.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[ 42%] Built target CollisionPrevention
[ 42%] Built target AirspeedValidator
[ 43%] Built target hysteresis
make[2]: *** [platforms/posix/CMakeFiles/gazebo.dir/rule] Error 2
make[1]: *** [gazebo] Error 2
make: *** [px4_sitl] Error 2

KumarSaurav366, 2020-06-14:
Please help, I tried installation process following :
https://dev.px4.io/v1.9.0/en/setup/dev_env_mac.html

Jaeyoung-Lim, 2020-06-14:
@KumarSaurav366 You don't have gazebo installed on your system, as written in the error log.

KumarSaurav366, 2020-06-14:
Thank you for letting me know, 
what is the significance of Xcode in the installation process? 
I have followed the gazebo installation from: 
http://gazebosim.org/tutorials?cat=install&tut=install_on_mac&ver=8.0
and after downloading this Error comes out:

Error: Your Xcode (10.3) is too outdated.
Please update to Xcode 11.4.1 (or delete it).
Xcode can be updated from the App Store.

So, is this installation problem is caused by the Xcode version? I am using homebrew for the installations

Jaeyoung-Lim, 2020-09-12:
@KumarSaurav366 The error log seems to state that is the case. Have you tried updating the Xcode version?

MaEtUgR, 2021-11-22:
Closing this as stale, please reopen if the issue persists.

closed 2021-11-22 by MaEtUgR