PX4/PX4-Autopilot #15109: mavlink custom_mode is not defined on common.xml

BazookaJoe1900, 2020-06-15:
I have noticed that one of the most important field of mavlink, the custom_mode that on the heartbeat is not really defined on the common.xml.
I can understand that custom_mode is custom per flight  stack, so might not be on common.xml. but still (in my opinion) it should be defined on a mavlink xml.

my first question do you agree?
if yes, should it enter into common.xml, or should we add it to new xml called px4.xml?

after it defined, the enum from the mavlink generated code should be used.
I can implement that after your notes.

@LorenzMeier ?

dagar, 2020-06-15:
This is a long standing problem, and frankly I think it's absurd that we waste 5 bytes in every single HEARTBEAT and still don't have a set of common modes.

I'm not in favor of introducing a PX4 specialization. It should be possible to have a common set of mavlink someone can target for compatibility beyond a single flight controller.

BazookaJoe1900, 2020-06-15:
are you sure you want to try and sync different flight stack, now?

dagar, 2020-06-15:
> are you sure you want to try and sync different flight stack, now?

In Mavlink every step we take should be towards something that's even possible to target as a shared standard across flight controllers, ground control stations, etc. Entrenching flight controller differences limits the value of Mavlink and ultimately hurts the entire ecosystem.

It's nearly impossible to build something "mavlink compatible" and have it work outside of the specific case you tested. I don't think it's too late to fix it.

hamishwillee, 2020-06-15:
Yes. 
- Ardupilot uses a dialect so publish their own sets - e.g. https://mavlink.io/en/messages/ardupilotmega.html#PLANE_MODE
- PX4 uses "common.xml" which means we can't do this (no dialect file in which to put them). Instead the definitions are in a private header. 

There has been some initial work on defining a common set [here](https://docs.google.com/document/d/1LIcfOL3JrX-EznvXArna1h-sZ7va7LRTteIUzISuD8c/edit). This did not yet resolve how to handle vehicle-specific peculariarities but there was broad agreement on a small set of common modes. 

This has somewhat stalled. Everyone agrees it is a useful and important thing to do, but there are other important thing that are considered "more" important.

BazookaJoe1900, 2020-06-16:
@hamishwillee 
I agree that there are more important thing to to, this is more of a code beautify than critical issue.
How about adding a new xml, called px4 that will define this enum, and maybe other that are not common?
this is easy, and not that dirty, solution.

hamishwillee, 2020-06-16:
PX4 only supports common code "as an ideology". I think we're better off continuing the work to define common modes.

BazookaJoe1900, 2021-07-04:
unstaling it...