PX4/PX4-Autopilot #15115: Disarm vehicle on kill switch

msyed123, 2020-06-15:
**Describe problem solved by the proposed feature**
Hello all. In this feature request, I am suggesting a modification to the vehicle flight behavior following the activation of a vehicle kill switch. I believe that this change may bring the functionality of a vehicle to be more inline with what an end-user would expect, although admittedly I am no expert on aerospace safety best practices. What I am observing in the latest release version of the firmware in Gazebo HITL and in real-life testing is the following:

1. Kill switch is triggered on the transmitter. Motors power down immediately.
2. Kill switch is disengaged. Vehicle state is resumed exactly as it was before the kill switch was pulled. Vehicle is therefore still armed, and motors spin up (provided that the user did not disarm while the kill switch was triggered.)

**Describe your preferred solution**
What I would suggest is that if a user pulls the kill switch, the vehicle flight mode should automatically switch to the disarmed flight mode. The reason I believe this is a better course of action is because I see the kill switch as a safety feature that allows the vehicle to be shutdown in the event of rogue or dangerous behavior. Following this behavior, the user should be able to have time to determine the cause of the failure that caused them to initiate the termination of the flight in the first place. Further, because the kill switch functionality has the ability to crash the drone while in mid-flight, it is very possible that the vehicle is no longer flyable after the fact and should therefore be disarmed even if the switch is no longer triggered.

**Describe possible alternatives**
_More aggressive disarming procedure:_ I don't even know if this exists but perhaps some description of a flight state that requires to be restarted before a flight.
_Less aggressive disarming procedure:_ Leave the functionality exactly as it is and perhaps leave a note in the documentation pertaining to the vehicle behavior on activation of the kill switch. 

**Additional context**
Thanks very much for taking the time to review my suggestion. I am very new to the stuff so please let me know if there is anything that I should do differently for one of these feature requests in the future.

dagar, 2020-06-16:
Current PX4 has auto-disarm after the kill switch is engaged for 5 seconds. Do you think that's too long?

https://github.com/PX4/Firmware/blob/39b803a9dc894cef7d086e4e699262424f4e1423/src/modules/commander/Commander.cpp#L1629-L1630

msyed123, 2020-06-16:
Ah, sorry, I didn't see that. Personally speaking, I expected the kill switch to disarm the drone immediately, and I would argue that it might be a better implementation. Still, I think I should have read the documentation better as well. Is there a reason that we wait for the switch to be held for some time before disarming?

dagar, 2020-06-16:
It was a compromise solution because some people seem to actually want it. You can configure it with `COM_KILL_DISARM`.

msyed123, 2020-06-16:
That makes sense. Thank you so much for the clarification and the time!

closed 2020-06-16 by msyed123