PX4/PX4-Autopilot #15129: Problems with PX4 SITL building on mac with protobuf 3.12

hanyang2013212, 2020-06-18:
What version of protobuf and what language are you using?
Version: 3.12.

What operating system (Linux, Windows, ...) and version?
Mac OS catalina

What did you do?
try tp build PX4 gazebo9 SITL on mac

$ make px4_sitl gazebo_uuv_hippocampus
What did you expect to see
I have tried on Ubuntu 18.04 and it's okay to build, but on Mac faild
What did you see instead?

In file included from MotorSpeed.pb.cc:4:
./MotorSpeed.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
./MotorSpeed.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
./MotorSpeed.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
./MotorSpeed.pb.h:40:46: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::google::protobuf::internal::AuxillaryParseTableField aux[];
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
                                             AuxiliaryParseTableField
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
In file included from MotorSpeed.pb.cc:4:
./MotorSpeed.pb.h:132:56: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedInputStream* input) final;
                                                       ^~~~~
./MotorSpeed.pb.h:134:64: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedOutputStream* output) const final;
                                                               ^~~~~
./MotorSpeed.pb.h:136:68: error: only virtual member functions can be marked 'final'
      bool deterministic, ::google::protobuf::uint8* target) const final;
                                                                   ^~~~~
./MotorSpeed.pb.h:174:33: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::google::protobuf::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
MotorSpeed.pb.cc:11:10: fatal error: 'google/protobuf/wire_format_lite_inl.h' file not found
#include <google/protobuf/wire_format_lite_inl.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
9 errors generated.

In file included from CommandMotorSpeed.pb.cc:4:
./CommandMotorSpeed.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
./CommandMotorSpeed.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
./CommandMotorSpeed.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
./CommandMotorSpeed.pb.h:40:46: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::google::protobuf::internal::AuxillaryParseTableField aux[];
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
                                             AuxiliaryParseTableField
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
In file included from CommandMotorSpeed.pb.cc:4:
./CommandMotorSpeed.pb.h:132:56: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedInputStream* input) final;
                                                       ^~~~~
[... 212 lines cut ...]
In file included from /Users/hanyang/Documents/Firmware/Tools/sitl_gazebo/src/gazebo_sonar_plugin.cpp:22:
In file included from /Users/hanyang/Documents/Firmware/Tools/sitl_gazebo/include/gazebo_sonar_plugin.h:38:
./Range.pb.h:133:56: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedInputStream* input) final;
                                                       ^~~~~
./Range.pb.h:135:64: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedOutputStream* output) const final;
                                                               ^~~~~
./Range.pb.h:137:68: error: only virtual member functions can be marked 'final'
      bool deterministic, ::google::protobuf::uint8* target) const final;
                                                                   ^~~~~
./Range.pb.h:234:33: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::google::protobuf::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
./Range.pb.h:268:16: error: use of undeclared identifier 'GOOGLE_LONGLONG'
  time_usec_ = GOOGLE_LONGLONG(0);
               ^
/Users/hanyang/Documents/Firmware/Tools/sitl_gazebo/src/gazebo_sonar_plugin.cpp:35:1: error: cannot initialize return object of type 'gazebo::SensorPlugin *' with an rvalue of type 'gazebo::SonarPlugin *'
GZ_REGISTER_SENSOR_PLUGIN(SonarPlugin)
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/common/Plugin.hh:445:12: note: expanded from macro 'GZ_REGISTER_SENSOR_PLUGIN'
    return new classname();\
           ^~~~~~~~~~~~~~~
10 errors generated.
Does anyone have the same issue with the building of SITL on mac?

julianoes, 2020-06-24:
Can you try `brew upgrade` and then `make clean` and `make px4_sitl gazebo` again?

julianoes, 2020-06-24:
Duplicate of #15091.

closed 2020-06-24 by julianoes