PX4/PX4-Autopilot #15132: Building emlid_navio2 error

fatbill27, 2020-06-18:
[0/1] Re-running CMake...
-- PX4 version: v1.11.0-beta2-215-gbe2081a2a2
-- PX4 config file: /home/fatbill27/Firmware/boards/emlid/navio2/default.cmake
-- PX4 config: emlid_navio2_default
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-313-ga296fe7
-- Configuring done
-- Generating done
-- Build files have been written to: /home/fatbill27/Firmware/build/emlid_navio2_default
[377/764] Building CXX object src/drivers/rc_input/CMakeFiles/drivers__rc_input.dir/RCInput.cpp.o
FAILED: src/drivers/rc_input/CMakeFiles/drivers__rc_input.dir/RCInput.cpp.o
/usr/bin/ccache /usr/bin/arm-linux-gnueabihf-g++  -DCONFIG_ARCH_BOARD_EMLID_NAVIO2 -DMODULE_NAME=\"rc_input\" -DPX4_MAIN=rc_input_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/emlid/navio2/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../boards/emlid/navio2/include -I../../platforms/posix/include -Iexternal/Install/include -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -std=gnu++14 -MD -MT src/drivers/rc_input/CMakeFiles/drivers__rc_input.dir/RCInput.cpp.o -MF src/drivers/rc_input/CMakeFiles/drivers__rc_input.dir/RCInput.cpp.o.d -o src/drivers/rc_input/CMakeFiles/drivers__rc_input.dir/RCInput.cpp.o -c ../../src/drivers/rc_input/RCInput.cpp
In file included from /usr/arm-linux-gnueabihf/include/string.h:495,
                 from ../../platforms/common/include/px4_platform_common/board_common.h:49,
                 from ../../boards/emlid/navio2/src/board_config.h:71,
                 from ../../src/drivers/rc_input/RCInput.hpp:38,
                 from ../../src/drivers/rc_input/RCInput.cpp:34:
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘RCInput::RCInput(const char*)’ at ../../src/drivers/rc_input/RCInput.cpp:63:10:
/usr/arm-linux-gnueabihf/include/bits/string_fortified.h:106:34: error: ‘char* __builtin_strncpy(char*, const char*, unsigned int)’ specified bound 20 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[382/764] Building CXX object src/modules/commander/CMakeFiles/modules__commander.dir/gyro_calibration.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:222: emlid_navio2] Error 1

dagar, 2020-06-18:
What's your build environment?

fatbill27, 2020-06-18:
ubuntu server 20.04 64bit
vmware 15

dagar, 2020-06-18:
> ubuntu server 20.04 64bit
> vmware 15

The build systems are still focused around Ubuntu 18.04, but we'll likely shift to 20.04 later this year. 

Options
1. fix the trivial errors or ignore the new warnings per module
    - eg add -Wno-error=stringop-truncation here
https://github.com/PX4/Firmware/blob/6afa7e4368c4bb925b298f2f50aa3c50cdebdf71/src/drivers/rc_input/CMakeLists.txt#L36
2. disable -Werror https://github.com/PX4/Firmware/blob/6afa7e4368c4bb925b298f2f50aa3c50cdebdf71/cmake/px4_add_common_flags.cmake#L71
3. build within docker (px4io/px4-dev-armhf): https://hub.docker.com/r/px4io/px4-dev-armhf
4. Use Ubuntu 18.04 for now (really it's the default compiler version you're changing)

If you're interested it would be helpful if you could fix the new warnings and open a pull request.

fatbill27, 2020-06-19:
thanks veru much
back to Ubuntu 18.04 and everythings work fine

closed 2020-06-19 by fatbill27