PX4/PX4-Autopilot #15133: gazebo_ros crashes when spawning the 3RD Iris drone

linuxsocke, 2020-06-18:
Hi, 
my gazebo_ros crashes when trying to spawn the Iris drone in both cases when adding manually and within a launch file. 

I was following the [steps](https://dev.px4.io/master/en/simulation/ros_interface.html) and tried (after successfully building):

`roslaunch px4 posix_sitl.launch`

At the very first time on execution I got this error:

.
.
.
[ INFO] [1592468746.064344087]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1592468746.066314047]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
X Error of failed request:  GLXBadDrawable
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  50
  Current serial number in output stream:  50
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)
[INFO] [1592468746.610718, 0.000000]: Loading model XML from file /home/linuxsocke/Projects/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[INFO] [1592468746.617037, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[gazebo-3] process has died [pid 14052, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/linuxsocke/Projects/Firmware/Tools/sitl_gazebo/worlds/empty.world 
...

When trying again I get:

.
.
.
[ INFO] [1592468988.816359241]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1592468988.834652782, 0.016000000]: Physics dynamic reconfigure ready.
[libprotobuf ERROR google/protobuf/descriptor_database.cc:57] File already exists in database: Imu.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1164] CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
Aborted (core dumped)
[gazebo-3] process has died [pid 15814, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/linuxsocke/Projects/Firmware/Tools/sitl_gazebo/worlds/empty.world 
...

I have to mention that I previously had problems with different versions of the protobuf compiler. I resolved this by using `make uninstall` to get rid of one version. I have now the latest version of protobuf from apt.

Jaeyoung-Lim, 2020-07-05:
@linuxsocke Can you try doing 
```
make distclean
make submodulesupdate
make px4_sitl gazebo
```
and then run the commands you are trying to run?

linuxsocke, 2020-07-06:
After executing **make px4_sitl gazebo** I get:

.
.
.
GAZEBO_PLUGIN_PATH :/home/linuxsocke/Projects/Firmware/build/px4_sitl_default/build_gazebo:/home/linuxsocke/Projects/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/linuxsocke/Projects/Firmware/Tools/sitl_gazebo/models:/home/linuxsocke/Projects/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /opt/intel/openvino_2020.2.130/data_processing/dl_streamer/lib:/opt/intel/openvino_2020.2.130/data_processing/gstreamer/lib:/opt/intel/openvino_2020.2.130/opencv/lib:/opt/intel/openvino_2020.2.130/deployment_tools/ngraph/lib:/opt/intel/openvino_2020.2.130/deployment_tools/inference_engine/external/hddl/lib:/opt/intel/openvino_2020.2.130/deployment_tools/inference_engine/external/gna/lib:/opt/intel/openvino_2020.2.130/deployment_tools/inference_engine/external/mkltiny_lnx/lib:/opt/intel/openvino_2020.2.130/deployment_tools/inference_engine/external/tbb/lib:/opt/intel/openvino_2020.2.130/deployment_tools/inference_engine/lib/intel64:/usr/local/cuda-10.0/lib64::/home/linuxsocke/Projects/ws_catkin/devel/lib:/opt/ros/melodic/lib:/home/linuxsocke/Projects/Firmware/build/px4_sitl_default/build_gazebo:/home/linuxsocke/Projects/Firmware/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
**[libprotobuf ERROR google/protobuf/descriptor_database.cc:57] File already exists in database: Imu.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1164] CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
SITL COMMAND: "/home/linuxsocke/Projects/Firmware/build/px4_sitl_default/bin/px4" "/home/linuxsocke/Projects/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/linuxsocke/Projects/Firmware"/test_data
INFO  [px4] Creating symlink /home/linuxsocke/Projects/Firmware/ROMFS/px4fmu_common -> /home/linuxsocke/Projects/Firmware/build/px4_sitl_default/tmp/rootfs/etc**

______  __   __    ___ 
|     ___ \ \ \  / /    /   |
|     |_/ /    \ V /   / /| |
|      __/   /    \   / /_| |
|  |         / /^\ \  \___  |
\_|          \/   \/       |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 1311244
  CAL_GYRO_PRIME: curr: 0 -> new: 1311244
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.5000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0022 -> new: 0.0050
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_P_ACC: curr: 4.0000 -> new: 12.0000
  MPC_Z_VEL_I_ACC: curr: 2.0000 -> new: 3.0000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P_ACC: curr: 1.8000 -> new: 4.0000
  MPC_XY_VEL_D_ACC: curr: 0.2000 -> new: 0.3200
  MPC_SPOOLUP_TIME: curr: 1.0000 -> new: 0.5000
  MPC_TKO_RAMP_T: curr: 3.0000 -> new: 1.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
Gazebo multi-robot simulator, version 9.13.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

  NAV_DLL_ACT: curr: 0 -> new: 2
[Msg] Waiting for master.
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000  
  RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
  SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[Err] [Connection.cc:546] Connection[0] Closed during Read
[Err] [ConnectionManager.cc:135] Unable to read from master
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo

linuxsocke, 2020-07-06:
I only can say that previously I had other protobuf versions due to some tensorflow stuff.

srmx, 2020-09-07:
I have a similar issue 
```
$ make px4_sitl gazebo_plane
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/px4.github/build/px4_sitl_default/...x4.github /home/px4.github/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/px4.github/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: plane
world: none
src_path: /home/px4.github
build_path: /home/px4.github/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/px4.github/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/px4.github/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/home/px4.github/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
[libprotobuf ERROR google/protobuf/descriptor_database.cc:57] File already exists in database: ignition/msgs/time.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1164] CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
/home/px4.github/Tools/sitl_run.sh: line 141: 15968 Aborted                 (core dumped) gzserver "${src_path}/Tools/sitl_gazebo/worlds/empty.world"
SITL COMMAND: "/home/px4.github/build/px4_sitl_default/bin/px4" "/home/px4.github"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/px4.github"/test_data
INFO  [px4] Creating symlink /home/px4.github/ROMFS/px4fmu_common -> /home/px4.github/build/px4_sitl_default/tmp/rootfs/etc
```
I tried the various versions of Gazebo and branches of PX4:
Ubuntu: 18.04
Gazebo: 11.0.0, 10.2.0
PX4: master, v1.11.0

linuxsocke, 2020-09-08:
I found out that this error come from some of the px4 gazebo plugins located in /Firmware/Tools/sitl_gazebo/src/. 

Those plugins are used in the robot description which either is in the .sdf or the .xacro format). When commenting out some of the plugins it works for me. But your robot is then missing the functionality of the plugins..

Jaeyoung-Lim, 2020-09-08:
@linuxsocke It sounds like you don't have your `GAZEBO_PLUGIN_PATH` properly. 

Try
```
cd <Firmware_clone>
DONT_RUN=1 make px4_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    # (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
```

linuxsocke, 2020-09-08:
My GAZEBO_PLUGIN_PATH is correct:
`GAZEBO_PLUGIN_PATH :/home/linuxsocke/Projects/Firmware/build/px4_sitl_default/build_gazebo:/home/linuxsocke/Projects/Firmware/build/px4_sitl_default/build_gazebo` 

Your plugins are loading but seem not to work due to the protobuf issue:
`[libprotobuf ERROR google/protobuf/descriptor_database.cc:57] File already exists in database: Imu.proto`

Jaeyoung-Lim, 2020-09-08:
@linuxsocke Have you tried a clean build in firmware? Usually that error happens when you have the protos generated but the definition changed

linuxsocke, 2020-09-08:
Yeah I tried that. The problem occurred on my old Ubuntu 18.04 system. 

In the past I thought the problem was due to a different protobuf version. Because when working with tf I followed [this](https://tensorflow-object-detection-api-tutorial.readthedocs.io/en/latest/install.html#protobuf-installation-compilation) steps.

Additionally, I added to my PATH in the .zshrc the following dir (when working with tf):  
`export PATH=/home/linuxsocke/Programs/Google_Protobuf/protoc-3.11.4-linux-x86_64${PATH:+:${PATH}}`

I removed that line later when working with px4 then. And when compiling the px4 stuff it told me in the output that it is using protobuf 3.0.0 (occurring like 3 times when creating the Makefile, dont know why it does it so often). Which is good then. 

So no idea what could resolve this. Its even quiet tough to reproduce this error :/ Maybe @srmx could help out with more info to reproduce this.

srmx, 2020-09-08:
Finally, I found the reason for the error. 
The first time the type `Imu.proto` is registered when loading `/usr/local/lib/libgazebo_msgs.so.11` (who knows why?). 
The second time it happens in `/home/px4.github/build/px4_sitl_default/build_gazebo/libgazebo_msgs.so.11` (as it must be).
Who knows what the damned version of `libgazebo_msgs.so.11` I had, but after removing it the issue left. 
For now `gzserver` sources the library `/usr/lib/x86_64-linux-gnu/libgazebo_msgs.so.11` instead.

closed 2020-09-09 by linuxsocke

s-niket, 2021-05-08:
Hi, 

I am facing similar issues. 

`[libprotobuf ERROR google/protobuf/descriptor_database.cc:57] File already exists in database: Imu.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1164] CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
Aborted (core dumped)
`

@msurov Can you describe the exact commands you used to solve this? Right now, I would uninstall gazebo and then reinstall it to have the simulation launch again. 

Thanks

Jaeyoung-Lim, 2021-05-09:
@s-niket It seems like you have a different `Imu.proto` already existing in your path, and your issue is not related.

Any chance you are running PX4 inside a catkin workspace?

msurov, 2021-05-09:
> Hi,
>
> I am facing similar issues.
>
> `[libprotobuf ERROR google/protobuf/descriptor_database.cc:57] File already exists in database: Imu.proto [libprotobuf FATAL google/protobuf/descriptor.cc:1164] [...]

As far as I remember, I removed manually all the gazebo libraries like /usr/local/lib/libgazebo_msgs.so.11 using the rm command. The usual reinstall does not do this.