PX4/PX4-Autopilot #15138: v1.11 beta VTOL standard issues (needs review)

dagar, 2020-06-18:
### Logs
1. https://logs.px4.io/plot_app?log=fc8a7b1e-24ec-4200-abe4-e76dc959047d log001_4_RC_test.ulg - we tested RC commands
2. https://logs.px4.io/plot_app?log=252166fd-5e22-4879-b8f5-9a0a053da57e log001_5_bad_takeoff_and_land.ulg - during take off VTOL was blown away because of the wind but he still manage to start mission, in vtol land he was also blown away and fall down on different location
3. https://logs.px4.io/plot_app?log=d1f696a9-e622-4d11-9821-64a80efa86ef log001_7_takeoff_failed.ulg - VTOL was blown away on take off, far far far away and he didn't start the mission, before touching the ground we tryed to save him with RC... when I came to him I saw that he was still running engings (he didn't disarm in the grass)

dagar, 2020-06-18:
These are logs from @MartinStokelj shared on slack I'm trying to capture in one place for review. Anything else about the vehicle you can share with us @MartinStokelj?

If you have previous good logs from the same vehicle it never hurts to upload (logs.px4.io) and share them for comparison.

MartinStokelj, 2020-06-19:
I hope this will be helpful.
VTOL - TC off
[VTOL4.zip](https://github.com/PX4/Firmware/files/4802681/VTOL4.zip)
VTOL- TC on, NO wind
[VTOL5.zip](https://github.com/PX4/Firmware/files/4802685/VTOL5.zip)
VTOL - TC on, wind, Weathervane bug on master
https://review.px4.io/plot_app?log=ed8bcc5a-08f8-4329-bb53-1f3a40f3fa03
https://review.px4.io/plot_app?log=26c014eb-74bb-46d9-9ffb-908480294b50
https://review.px4.io/plot_app?log=b858d7d3-c0ea-4e76-bed5-b301ada47f98
https://review.px4.io/plot_app?log=3ffcdc77-0c76-4d73-8c33-758e31ce1938
https://review.px4.io/plot_app?log=19913b92-33c1-410c-a030-bcf97dc9762c

https://review.px4.io/plot_app?log=24797ba1-2a7f-437a-ac5a-fc40bec0d4e8
https://review.px4.io/plot_app?log=2d2fb371-a815-4ce9-af8b-94d0db90b258
https://review.px4.io/plot_app?log=33288158-d40f-4e86-92e8-02d4440b576f
https://review.px4.io/plot_app?log=596ca957-de8d-4026-9cf2-5f2bd45f51cc
https://review.px4.io/plot_app?log=482d3f38-ca3c-4f0e-a209-e4cc0636e870

MartinStokelj, 2020-06-19:
Perhaps it would be good to check parameters. Allowing yaw in vtol takeoff?
[param_VTOL.zip](https://github.com/PX4/Firmware/files/4802885/param_VTOL.zip)

sfuhrer, 2020-06-19:
@MartinStokelj are you using a DeltaQuad or a custom airframe? I think the hover thrust margin is really low, the vehicle needs about 80% to fly level, right? Or the other way round, the vehicle sometimes barely climbs when full on throttle.
[image]
I personally haven't seen a VTOL fly well that had a hover thrust above 60%, as on a VTOL (especially in wind) the MC motors have to fight quite a bit against the aerodynamic forces and moments generated by the body.

MartinStokelj, 2020-06-19:
@sfuhrer we are using DeltaQuad. So what do you suggest?

MartinStokelj, 2020-07-01:
We increased the propellers and make new flights. Wind was low today.
RC test 
https://review.px4.io/plot_app?log=37caf131-19a8-42db-b9fb-bc06677de3c6
2x flight (great VTOL take off, bad VTOL land)
https://review.px4.io/plot_app?log=9bcf5d46-7479-4841-9049-f2f2e1c330dd
https://review.px4.io/plot_app?log=d6a593c1-a715-41c4-8a99-7a68cf4b7045
Crash (transition timeout has forced him to go 100agl then we lose control in manual mode (probably one propeller fall off since VTOL start to spinning around (yaw)))
https://review.px4.io/plot_app?log=2a8bea0b-7148-4202-a516-e5cdc8b8b165

sfuhrer, 2020-07-08:
@MartinStokelj sorry to hear about the crash. Hope it's not too bad. 
You can increase the VTOL transition timeout via `VT_TRANS_TIMEOUT`, but actually it's pretty bad if the vehicle can't accelerate within 15s to cruise airspeed, so I would also there consider upgrading your motor/propeller.
Let me know if I can help you with anything else specifically.

MartinStokelj, 2020-07-14:
Today we had wind and went flying again. We actually made only 1 flight but even that one had problem in transition.

no global position on VTOL take off, VTOL has landed, I beleave he even shut down motors in air.
https://review.px4.io/plot_app?log=16c22dbe-c8da-4e5f-842f-8e364b6215af
https://review.px4.io/plot_app?log=9f74123d-2118-499b-a463-37d90c9f8b5c
We did calibrations again.
Actual flight (AMSL problem when doing transition to fixed wing, Fused altitude is way off but only in transition time, we saw negative AMSL and AGL in QgroundControl)
https://review.px4.io/plot_app?log=8b63304d-c5b3-4941-bc83-26b0f316a69d
VTOL Takeoff (AMSL problem when doing transition to fixed wing, Fused altitude is way off all the time of the flight)
https://review.px4.io/plot_app?log=e0dc1e82-3ef9-44fb-a2e1-5f0ebd09a3fe
https://review.px4.io/plot_app?log=df746d69-510e-4d19-8167-e63442185bf5

Extra question: Is it possible to show throttle % in QgroundControl?

MartinStokelj, 2020-07-15:
@ThomasRigi commented on slack "I think the increased vibrations during transitions throw off your accelerometer sensors"

Since last crash we gave same size propellers but they are very soft, we will buy the previous ones again if this is doing the vibrations. We also ordered more stronger motor as @sfuhrer suggested. 
I am a bit shocked that vibrations can affect the calculated altitude. PX4 should detect that fused altitude is way off from GPS and Barometer altitude.

LorenzMeier, 2021-01-02:
I'm closing this issue as stale - @MartinStokelj on your comment on altitude: The hard part in system engineering is to detect failure and to not detect a "sane" system as faulty at the same time. Post-flight that's easy, but building a reliable algorithm is an art. A decade of work has gone into the state estimation framework and it is extremely robust for reasonable setups (and it is deployed in tens of thousands of products) - but there are limits to everything.

closed 2021-01-02 by LorenzMeier