PX4/PX4-Autopilot #15141: Ground Effect Detection should use height above home

dlwalter, 2020-06-18:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
Currently, if there is no range finder on the aircraft, it assumes it is in ground effect whenever it is descending.  We are running into this issue balancing landing detection and altitude tracking.  Since the code assumes it is in ground effect while descending, negative Baro innovations are ignored, which causes height estimates to diverge and results in poor altitude tracking during landing.  

When the aircraft has a range finder it only assumes ground effect when [EKF2_GND_EFF_MAX_HGT above the ground](https://github.com/PX4/Firmware/blob/7aac0c5ac384cd25e2e6bce07920b875e476b668/src/modules/ekf2/ekf2_main.cpp#L1335) otherwise that parameter is ignored.  Instead the aircraft relies on land_detector to determine the ground effect - which will respond true if the aircraft is descending and not moving horizontally. 

**Describe your preferred solution**
**I think that the aircraft should use the height above home when determining if in ground effect.**  This is currently how MPC_LAND_ALT1 and MPC_LAND_ALT2 are both implemented using dist_to_ground.

There is a risk when not landing from your takeoff point where terrain is at a higher altitude.  

With the current code the aircraft won't use MPC_LAND_ALT1 and MPC_LAND_ALT2 and will end up landing at MPC_Z_VEL_MAX_DN - which could be a very hard landing.  In addition the current implementation will result in poor altitude tracking and the aircraft will think it is higher than it actually is (since it is ignore negative Baro innovations during the descent).  

It would be possible to only implement the above change when within x meters from home.  Use the current implementation when > x meters horizontally from home.

There could be some discussion had on if landing away from your takeoff point is an edge case for the majority of users.  If you are doing so without a range finder at a higher altitude then automated landings will be pretty poor  anyways due to the aircraft ignoring landing speeds.

dagar, 2020-06-18:
Yesterday on the dev call we briefly mentioned the idea of things like MPC_LAND_ALT{1,2}  only applying to altitude above home when actually close to home (if you don't have a distance sensor). @MaEtUgR

dlwalter, 2020-06-18:
If we ignore MPC_LAND_ALT{1,2} when far enough from home, then is the plan to just let the aircraft descend at full speed MPC_Z_VEL_MAX_DN?  

Always using MPC_LAND_ALT:
-If you are on a hill you end up with conservative descent speeds.
-If you are in a bowl you end up fast descent speeds

Ignoring MPC_LAND_ALT far from home:
-If you are on a hill you end up with fast descent speeds.
-If you are in a bowl you end up fast descent speeds

I guess the risk is if your MPC_Z_VEL_MAX_DN is too fast for a safe autonomous landing.

dagar, 2020-06-18:
> If we ignore MPC_LAND_ALT{1,2} when far enough from home, then is the plan to just let the aircraft descend at full speed MPC_Z_VEL_MAX_DN?

No real plan yet, pedantically working through the details like this is exactly what needs to happen to get to the best (least bad?) option.

I mainly want to remove anything potentially unintuitive, and here I think that means only using that logic when you're sure. That also means an actual commanded LAND away from HOME with no distance sensor would have to default to a safe descent speed all the way down. Having a set landing waypoint could be another exception where the altitude of that point is respected.

We could also try to make the behavior a lot clearer in different situations and subtly show just how valuable a distance sensor can be.

dlwalter, 2020-06-18:
That makes sense.  I would then hope that ekf2 would use the ground effect parameter in the same manner - then autonomous landings away from home would be conservative with ground effect while landings near home would only assume ground effect below EKF2_GND_EFF_MAX_HGT.

dagar, 2020-06-18:
### Action Items
1. Only ever use altitude above home in the vicinity of home (kind of sounds tautological...). Cylinder or cone?
    - review MPC_LAND_ALT{1,2}, MPC_LAND_ALT, etc
2. `MulticopterLandDetector::_get_ground_effect_state()` improvements
https://github.com/PX4/Firmware/blob/0a63df25a63c683fc849c2431ccee472a3716bfa/src/modules/land_detector/MulticopterLandDetector.cpp#L308-L311
     - make use of altitude above home when near home?
     - use distance to ground if available?
     - this part is probably a bigger discussion with @RomanBapst

dagar, 2020-06-18:
Actually, distance to ground is already included. https://github.com/PX4/Firmware/blob/0a63df25a63c683fc849c2431ccee472a3716bfa/src/modules/ekf2/ekf2_main.cpp#L1331-L1344

dlwalter, 2020-06-18:
Yep, it's right there, but won't use it because dist_bottom_valid is false without terrain data.

dlwalter, 2020-06-18:
Modifying line 1334 with an OR on the distance from home would be an easy fix here.

dagar, 2020-06-18:
> Modifying line 1334 with an OR on the distance from home would be an easy fix here.

That seems pretty inoffensive to me. Do you want to give it a try and if it seems viable raise a PR to discuss with @RomanBapst, @priseborough, @bresch?

dlwalter, 2020-06-18:
Yeah, I can start on a PR.  It doesn't really cover the MPC_LAND_ALT{1,2} issue - land_detector logic may also need some work.

closed 2021-02-03 by LorenzMeier