PX4/PX4-Autopilot #15146: intermittent SITL test failures

dagar, 2020-06-19:
https://github.com/PX4/Firmware/pull/15136/checks?check_run_id=788640705

[image]

dagar, 2020-06-19:
Is this time jump the problem?

[image]

dagar, 2020-06-20:
https://github.com/PX4/Firmware/pull/15154/checks?check_run_id=790699473

[image]

It looks like the Iris dual GPS model is losing global position for some reason.

dagar, 2020-06-23:
https://github.com/PX4/Firmware/pull/15198/checks?check_run_id=799637423

[image]

Jaeyoung-Lim, 2020-06-23:
@dagar Should we maybe take it out for now?

TSC21, 2020-06-23:
> @dagar Should we maybe take it out for now?

To remove this one we would have to remove other as well, since the intermitent failures happen with other models as well. I have run this specific test several times straight locally without a single error, so this needs to be tracked in a different matter.

dagar, 2020-06-23:
Can someone take a look at the logs in detail? If there isn't already enough data captured the builds need to evolve so that it's possible to debug when these intermittent failures inevitably occur. Being able to reproduce it locally is going to get harder, so ultimately shouldn't be that different than reviewing a real flight log.

Jaeyoung-Lim, 2020-06-23:
The link no longer shows the result of the failure...

dagar, 2020-06-23:
> The link no longer shows the result of the failure...

Which?

> It would be nice if we can get the gazebo logs separately, since currently the logs are not visible. Is there already a way to upload artifacts?

Yes, we can archive whatever we want, just don't do it by default on the successful builds to save space.

Jaeyoung-Lim, 2020-06-23:
> > The link no longer shows the result of the failure...

>Which?

https://github.com/PX4/Firmware/pull/15136/checks This one. I just realized the other link shows the error correctly.

julianoes, 2020-06-24:
I ran it locally and the dual_gps test failed first time for me:
```
[gzserver    ] [Wrn] [gazebo_gps_plugin.cpp:76] [gazebo_gps_plugin]: iris_dual_gps::gps1 using gps topic "gps1"
[gzserver    ] [Wrn] [gazebo_gps_plugin.cpp:201] [gazebo_gps_plugin] Using default update rate of 5hz 
[gzserver    ] [Wrn] [gazebo_gps_plugin.cpp:76] [gazebo_gps_plugin]: iris_dual_gps::gps0 using gps topic "gps0"
[gzserver    ] [Wrn] [gazebo_gps_plugin.cpp:201] [gazebo_gps_plugin] Using default update rate of 5hz 
[mavsdk_tests] Filters: Offboard takeoff and land
[mavsdk_tests] [03:23:48|Info ] MAVSDK version: 0.27.0-30-g7dd51855 (mavsdk_impl.cpp:26)
[mavsdk_tests] Waiting for system connect
[mavsdk_tests] [03:23:48|Info ] New system on: 127.0.0.1:14580 (udp_connection.cpp:256)
[mavsdk_tests] [03:23:48|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:372)
[mavsdk_tests] [03:23:48|Debug] Component Autopilot (1) added. (system_impl.cpp:352)
[mavsdk_tests] [03:23:48|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:525)
[mavsdk_tests] Waiting for system to be ready
[mavsdk_tests] Waiting to get home position
[px4         ] INFO  [commander] ARMED by Arm/Disarm component command
[px4         ] WARN  [mc_pos_control] Offboard activation failed with error: Activation Failed
[px4         ] WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
[px4         ] WARN  [mc_pos_control] Altitude-Ctrl activation failed with error: Activation Failed
[px4         ] INFO  [commander] Takeoff detected
[px4         ] INFO  [ecl/EKF] 139084000: reset velocity to GPS
[px4         ] INFO  [ecl/EKF] 139084000: reset position to GPS
[px4         ] INFO  [ecl/EKF] 139084000: Emergency yaw reset - mag use stopped
[px4         ] WARN  [commander] Stopping compass use! Check calibration on landing
[mavsdk_tests] [03:23:54|Debug] MAVLink: info: ARMED by Arm/Disarm component command (system_impl.cpp:257)
[mavsdk_tests] [03:23:54|Debug] MAVLink: info: Takeoff detected (system_impl.cpp:257)
[mavsdk_tests] [03:23:54|Debug] MAVLink: critical: Stopping compass use! Check calibration on landing (system_impl.cpp:257)
```

```
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Offboard takeoff and land
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] /home/julianoes/src/Firmware/test/mavsdk_tests/test_multicopter_offboard.cpp:42
[mavsdk_tests] ...............................................................................
[mavsdk_tests] /home/julianoes/src/Firmware/test/mavsdk_tests/autopilot_tester.cpp:219: FAILED:
[mavsdk_tests]   CHECK( poll_condition_with_timeout( [ = ]() { return estimated_position_close_to(target, acceptance_radius_m); }
[mavsdk_tests] with expansion:
[mavsdk_tests]   false
[mavsdk_tests] Target position reached
```

https://logs.px4.io/plot_app?log=8c840c92-3c83-45ee-94ad-21d8657ec161

julianoes, 2020-06-24:
And another one:

```
[mavsdk_tests] Filters: Offboard takeoff and land
[mavsdk_tests] [03:33:50|Info ] MAVSDK version: 0.27.0-30-g7dd51855 (mavsdk_impl.cpp:26)
[mavsdk_tests] Waiting for system connect
[mavsdk_tests] [03:33:50|Info ] New system on: 127.0.0.1:14580 (udp_connection.cpp:256)
[mavsdk_tests] [03:33:50|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:372)
[mavsdk_tests] [03:33:50|Debug] Component Autopilot (1) added. (system_impl.cpp:352)
[mavsdk_tests] [03:33:50|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:525)
[mavsdk_tests] Waiting for system to be ready
[mavsdk_tests] Waiting to get home position
[px4         ] INFO  [commander] ARMED by Arm/Disarm component command
[px4         ] WARN  [mc_pos_control] Offboard activation failed with error: Activation Failed
[px4         ] WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
[px4         ] WARN  [mc_pos_control] Altitude-Ctrl activation failed with error: Activation Failed
[px4         ] INFO  [commander] Takeoff detected
[px4         ] ERROR [commander] Critical failure detected: lockdown
[mavsdk_tests] [03:33:56|Debug] MAVLink: info: ARMED by Arm/Disarm component command (system_impl.cpp:257)
[mavsdk_tests] [03:33:56|Debug] MAVLink: info: Takeoff detected (system_impl.cpp:257)
[mavsdk_tests] [03:33:56|Debug] MAVLink: emergency: Critical failure detected: lockdown (system_impl.cpp:257)
[mavsdk_tests] [03:33:56|Warn ] Unknown StatusText severity (telemetry_impl.cpp:891)
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[mavsdk_tests] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[mavsdk_tests] mavsdk_tests is a Catch v2.9.2 host application.
[mavsdk_tests] Run with -? for options
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Offboard takeoff and land
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] /home/julianoes/src/Firmware/test/mavsdk_tests/test_multicopter_offboard.cpp:42
[mavsdk_tests] ...............................................................................
[mavsdk_tests] /home/julianoes/src/Firmware/test/mavsdk_tests/autopilot_tester.cpp:219: FAILED:
[mavsdk_tests]   CHECK( poll_condition_with_timeout( [ = ]() { return estimated_position_close_to(target, acceptance_radius_m); }, timeout_duration) )
[mavsdk_tests] with expansion:
[mavsdk_tests]   false
[mavsdk_tests] Target position reached
[px4         ] INFO  [commander] Landing detected
[px4         ] INFO  [commander] DISARMED by Auto disarm initiated
[px4         ] INFO  [logger] closed logfile, bytes written: 28312745
[mavsdk_tests] [03:34:18|Debug] MAVLink: info: Landing detected (system_impl.cpp:257)
[mavsdk_tests] [03:34:18|Debug] MAVLink: info: DISARMED by Auto disarm initiated (system_impl.cpp:257)
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[mavsdk_tests] target_pos.lat: 47.3978
[mavsdk_tests] target_pos.lon: 8.54561
[mavsdk_tests] current.lat: 47.3978
[mavsdk_tests] current.lon: 8.54558
[mavsdk_tests] Distance: 2.33083
[mavsdk_tests] Acceptance radius: 1
[mavsdk_tests] /home/julianoes/src/Firmware/test/mavsdk_tests/autopilot_tester.cpp:91: FAILED:
[mavsdk_tests]   CHECK( ground_truth_horizontal_position_close_to(_home, acceptance_radius_m) )
[mavsdk_tests] with expansion:
[mavsdk_tests]   false
[mavsdk_tests] ===============================================================================
[mavsdk_tests] test cases:  1 |  0 passed | 1 failed
[mavsdk_tests] assertions: 14 | 12 passed | 2 failed
[px4         ] INFO  [commander] Failsafe mode activated
[px4         ] INFO  [navigator] RTL HOME activated
[px4         ] PX4 Exiting...
[px4         ] Exiting NOW.
```

https://logs.px4.io/plot_app?log=10f13430-6ba4-418b-a740-a0743e5d0c57

julianoes, 2020-06-24:
`opt_flow_mockup` fails as well for me:

```
[mavsdk_tests] Filters: Offboard takeoff and land
[mavsdk_tests] [03:41:02|Info ] MAVSDK version: 0.27.0-30-g7dd51855 (mavsdk_impl.cpp:26)
[mavsdk_tests] Waiting for system connect
[mavsdk_tests] [03:41:02|Info ] New system on: 127.0.0.1:14580 (udp_connection.cpp:256)
[mavsdk_tests] [03:41:02|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:372)
[mavsdk_tests] [03:41:02|Debug] Component Autopilot (1) added. (system_impl.cpp:352)
[mavsdk_tests] [03:41:02|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:525)
[mavsdk_tests] Waiting for system to be ready
[mavsdk_tests] Waiting to get home position
[px4         ] INFO  [commander] ARMED by Arm/Disarm component command
[px4         ] WARN  [commander] Failsafe enabled: no RC
[px4         ] INFO  [commander] Failsafe mode activated
[px4         ] INFO  [commander] Failsafe mode deactivated
[px4         ] WARN  [mc_pos_control] Offboard activation failed with error: Activation Failed
[px4         ] WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
[px4         ] WARN  [mc_pos_control] Altitude-Ctrl activation failed with error: Activation Failed
[px4         ] INFO  [commander] Takeoff detected
[px4         ] ERROR [commander] Critical failure detected: lockdown
[mavsdk_tests] [03:41:08|Debug] MAVLink: info: ARMED by Arm/Disarm component command (system_impl.cpp:257)
[mavsdk_tests] [03:41:08|Debug] MAVLink: critical: Failsafe enabled: no RC (system_impl.cpp:257)
[mavsdk_tests] [03:41:08|Debug] MAVLink: info: Failsafe mode activated (system_impl.cpp:257)
[mavsdk_tests] [03:41:08|Debug] MAVLink: info: Failsafe mode deactivated (system_impl.cpp:257)
[mavsdk_tests] [03:41:08|Debug] MAVLink: info: Takeoff detected (system_impl.cpp:257)
[px4         ] WARN  [mc_pos_control] PositionControl: invalid setpoints
[px4         ] WARN  [mc_pos_control] Failsafe: blind land
[px4         ] INFO  [commander] Landing detected
[px4         ] INFO  [commander] DISARMED by Auto disarm initiated
[px4         ] INFO  [logger] closed logfile, bytes written: 6847405
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[px4         ] ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[mavsdk_tests] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[mavsdk_tests] mavsdk_tests is a Catch v2.9.2 host application.
[mavsdk_tests] Run with -? for options
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Offboard takeoff and land
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] /home/julianoes/src/Firmware/test/mavsdk_tests/test_multicopter_offboard.cpp:42
[mavsdk_tests] ...............................................................................
[mavsdk_tests] /home/julianoes/src/Firmware/test/mavsdk_tests/autopilot_tester.cpp:219: FAILED:
[mavsdk_tests]   CHECK( poll_condition_with_timeout( [ = ]() { return estimated_position_close_to(target, acceptance_radius_m); }, timeout_duration) )
[mavsdk_tests] with expansion:
[mavsdk_tests]   false
[mavsdk_tests] Target position reached
[mavsdk_tests] ===============================================================================
[mavsdk_tests] test cases:  1 |  0 passed | 1 failed
[mavsdk_tests] assertions: 14 | 13 passed | 1 failed
```

https://logs.px4.io/plot_app?log=b3caa0ad-fa17-4e2b-8fd0-1487cfeec681

dagar, 2020-06-24:
Another one from master last night. https://github.com/PX4/Firmware/runs/802554839

[image]

PR #15213 (2020-06-24): Remove flaky tests
Trying to fix #15146.
closed 2020-06-24 by dagar