PX4/PX4-Autopilot #15153: Implicit conversion int32_t->uint16_t leads to "ERROR: wp index out of bounds"

Roman-, 2020-06-20:
When sending missions that fail to pass the feasibility check (e.g. "Altitude too low"), PX4 sends one useful statustext, but also a bunch of "ERROR: wp index out of bounds" from [here](https://github.com/PX4/Firmware/blob/v1.10.2/src/modules/mavlink/mavlink_mission.cpp#L281).

What causes this is setting `_transfer_current_seq` to -1 either [there](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L922) or [there](https://github.com/PX4/Firmware/blob/v1.10.2/src/modules/mavlink/mavlink_mission.h#L122), and then passing this "-1" value through a bunch of int32_t variables to finally pass it to `send_mission_current()` which accpets seq number as `uint16_t`. That makes -1 become 65535 and causes seq number to be greater than item_count:

https://github.com/PX4/Firmware/blob/v1.10.2/src/modules/mavlink/mavlink_mission.cpp#L268

I'm referencing v1.10.2 but these parts of the code haven't been changed today. Sending missions via mavros in Gazebo simulator.

dagar, 2020-06-20:
Good catch, could you give https://github.com/PX4/Firmware/pull/15156 a try?

Roman-, 2020-06-21:
> Good catch, could you give #15156 a try?

Yes, now it doesn't throw these statustexts. Thanks

closed 2020-06-21 by dagar