PX4/PX4-Autopilot #15155: V1.11.0 (Beta&Master) MC control doesn´t switch off in FW (tiltrotor-vtol)

taileron, 2020-06-20:
all mc mot control remains active (VT_FW_MOT_OFFID doesn´t apply) even the nacelles are full swivelled to FW (differental thrust is only added in FW)

It can be testet on the ground with Convergence Vtol: 
arm -> apply a bit throttle -> switch to FW -> nacelles swivel in 2 steps fully forward to FW -> Mot 3 continues running -> all 3 Mot are still under MC control if you apply roll/pitch commands

fully to FW ->In this case Mot 3 should turn off and Mot 1,2 now should only be controlled by differential thrust 

https://review.px4.io/plot_app?log=dcd387e6-8402-4b7b-b53c-14096527066f

Convergence with Pixracer R14

sfuhrer, 2020-06-21:
I was able to reproduce the issue with master. It only happens in Acro flight mode, e.g. in stabilized the rear motor switches off correctly.

taileron, 2020-06-21:
There are always 3 errors necessary for a crash ... I have increased the Pitch PID extremely for testing High Alpha flight and strong vibrations consume extremely high currents from the servo supply so the left tilt servo finally was stuck at the bottom. With jigsaw puzzle technique the plane is almost back in one piece ...

taileron, 2020-06-22:
The plane is now pieced together again.
The problem also appears in other modes than Acro as far as I could test it on the ground.

sfuhrer, 2020-06-22:
@taileron it was working for me in e.g. stabilized. Could you share a log where you try it in stabilized and it is not working?

taileron, 2020-06-22:
https://review.px4.io/plot_app?log=5348e467-da1e-41af-90f5-359b7bbc4a43

taileron, 2020-06-23:
https://review.px4.io/plot_app?log=d7027b05-d53f-4668-b5dd-948cb27997ad

Stabilized in transition phase mot 3 stops for a moment. Afterwards mc pid are decoupled but pure throttle still controls mot 3 so it runs further with a bit lower rpm

taileron, 2020-07-04:
in the meantime I have found the cause of the crash: it was the somewhat inaccurate servo travel limitation of the tilt servos due to the pitch oscillation the right nacelle was caught at the lowest point above the apex of the control arm and could therefore no more return to the neutral point.
-> PWM Limits should be freely set without changing other aspects of servo and mixer functions ....

sfuhrer, 2020-07-08:
@taileron I'm still not sure what is causing the different behavior on your vehicle compared to mine. 
Could you check if anything changes when you set `PWM_MIN` and `PWM_MAX` to default? While in MC, the PWM on the MC are set to `VT_IDLE_PWM_MC`.

taileron, 2020-07-08:
@sfuhrer I´ll test it ... a few issues are related to the different approach of boards that come with or without io-board in my case it was an only FMU board (PixRacer) my other boards (e.g. PixFalcon) that have an io-board still don´t talk with Spectrum satellites as long as the io-board is activated so I cannot test it there. IO-board disabled (SYS_USE_IO=0) causes only 6 pwm outs and same behaviour

sfuhrer, 2020-07-08:
Aha okay. Then maybe also see if 'VT_MC_ON_FMU' to 1 does make any difference.

taileron, 2020-07-08:
https://review.px4.io/plot_app?log=9b8f11b7-a0a3-470b-8d16-faea008decdc
VT_MC_ON_FMU 0 or 1 behaves identical 
PWM limits now on defaults
SYS_USE_IO=1 causes manual control lost

sfuhrer, 2020-07-09:
@taileron I've tried to reproduce your issue by loading the default config of the Congervence on a Pixhawk4, and then set SYS_USE_IO to 0. It then still worked as expected:
[image]

https://review.px4.io/plot_app?log=1bece002-bc4a-481e-a97a-71d52cba69d0
So it looks a bit like it could really be a board issue.

taileron, 2020-07-09:
Tried it already on 2 different boards with the same result (PixFalcon and Pixracer)
Could there be an additional parameter which I have overlooked.
It is strange that in FW mode pure throttle is always mixed to all motors.
And it is strange as well that if the io-board ist active no rc inputs were processed anymore.

that´s a log from PixRacer built into Convergence (with vectored thrust instead of elevon control cause it offers only 6 pwm outs) 
https://review.px4.io/plot_app?log=df6c7f49-d3e0-4373-9e13-bc48a8c9c85e

taileron, 2020-07-11:
in one of 25 start procedures the DSMX receiver is recognized despite activated IO module. Then it does not show vtol status in the log, but Mot 3 now behaves correctly and goes out at FW even in Acro mode.
https://review.px4.io/plot_app?log=e9c4236a-4024-4ce6-8d24-35e5199bcca8
So the board behave different with io-board activated or not.

As soon as io-board is disabled, the 6 outputs of Pixracer and Pixfalcon both behave incorrectly with respect to motor 3 remaining on at FW and the receiver is always reliably detected.

taileron, 2020-07-26:
@sfuhrer after the solution for #15003 I could do more detailed tests. 
You are right, with io present it only doesn´t work immediately in acro mode, but once activated FW stability mode, everything in FW acro will work as well. (Mot 3 off and MC PID control off). Only then the Yaw I and other values are applied correctly. There are still small errors in the logical Vtol sequences.

With IO off or with all my FMU boards without IO unfortunately in FW Mot 3 always stays on and Mot 1,2 is under constant MC control so that it is still not flyable.

LorenzMeier, 2021-01-02:
Is this still an issue?

taileron, 2021-01-05:
@LorenzMeier still all the same behaviour: on FMU only fcs (e.g. Pixracer) mot 3 doesn´t switch off at all in fw mode and mc control remains always active in fw mode.
The Limit of the mot_offid corresponding output isn´t shifting without IO.
Boards with io behave o.k. if acro mode isn´t used after transition to fw.
In acro, mc control remains also active for the remaining outputs with io but since the limit has been moved properly down,  mot 3 is out. 
The shifted limit of mot 3 in fw can be always showed with pwm info.
I need to fly in acro after transition to fw to be able for further research in vectored thrust.
So far I have not been able to find this limit shift in the code.

taileron, 2021-09-29:
In the meantime, I was able to go a little deeper. The limit shift works over many release up to the current master on FC with IO or FMU with 8 outputs (such as mro control zero / H7) quite smoothly (Mot 3 goes off in FW as intended). 
The problem occurs only on FC without IO with 6 pwm outs. e.g. PixRacer or PixFalcon (with switched off IO) also across all firmware versions.