PX4/PX4-Autopilot #15168: Documentation Update Request: wrong topic in external vision odometry

marcelino-pensa, 2020-06-20:
### Issue details
I am using Px4 1.10 with Mavros 1.2.0. I was following the documentation [from here](https://dev.px4.io/v1.10/en/ros/external_position_estimation.html) to feed odometry to px4. It took me a while to realize that the documentation was wrong, as I have to publish into the topic `/mavros/odometry/out` instead of the documented one (`/mavros/odometry/odom`):

[image]

We can find the correct topics in `mavros_extras/src/plugins/odom.cpp`.

### MAVROS version and platform

Mavros: 1.2.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: 1.10.0

julianoes, 2020-06-24:
FYI @TSC21

TSC21, 2020-06-24:
This overall also needs an update on the frames of reference as well. @marcelino-pensa I see you created a commit on your repo. You mind creating a PR to the Devguide?

marcelino-pensa, 2020-06-24:
@TSC21 , I have just updated the documentation (see [this PR](https://github.com/PX4/Devguide/pull/1051))
Please leave comments if you believe that the updated descriptions are not good enough

marcelino-pensa, 2020-07-03:
I have another question regarding the documentation (and I should probably reflect the answer into the documentation).
If we are fusing odometry and the odometry source is not at the drone center, then the measured velocities are a mix of linear velocities and rotating velocities, something like the equation below, where \Omega is the drone's angular velocity and \bm{r} is the distance between the drone's center of mass and the odometry source:
[image]

What we need to feed px4 is the linear velocity portion, so one option would be to subtract the [image] from the measured velocities. 

On the other hand, px4 has the parameters `EKF2_EV_POS_X`, `EKF2_EV_POS_Y`, `EKF2_EV_POS_Z`... 
My question is: does Px4 compensate for the gyroscopic terms in the velocity measurement, or do we have to first eliminate the [image] terms, regardless of the values within `EKF2_EV_POS_X`, `EKF2_EV_POS_Y`, `EKF2_EV_POS_Z`?

marcelino-pensa, 2020-07-09:
Since my original question was already answered and merged into https://github.com/PX4/Devguide/pull/1051, should I create a different issue for my question above? @julianoes , @TSC21

hamishwillee, 2020-07-15:
@TSC21 Ping ^^^^

TSC21, 2020-07-15:
> I have another question regarding the documentation (and I should probably reflect the answer into the documentation).
> If we are fusing odometry and the odometry source is not at the drone center, then the measured velocities are a mix of linear velocities and rotating velocities, something like the equation below, where [image] is the drone's angular velocity and [image] is the distance between the drone's center of mass and the odometry source:

If you are doing that then you are not respecting the convention for the [`ODOMETRY`](https://mavlink.io/en/messages/common.html#ODOMETRY) message, which respects the [ROS REP 147 standard for aerial vehicles](http://www.ros.org/reps/rep-0147.html).

TSC21, 2020-07-15:
Closing this as the original issue was solved. @marcelino-pensa if you have further questions please bring them to the #ros channel in the PX4 Slack.

closed 2020-07-15 by TSC21

marcelino-pensa, 2020-07-15:
@TSC21 , I would disagree that ROS specifies twist as the origin with no offset. It has to be in body frame, but not at the COM (or drone center) necessarily:
```
The twist in this message should be specified in the coordinate frame given by the child_frame_id
```

In any case, I think that you answered my question. From your answer it sound like EKF2 does not take into account these offsets when estimating velocity, although it does when estimating position (counter-intuitive to me).
Since this is not clear in the documentation, I will create another issue and update the documentation to clarify this.