PX4/PX4-Autopilot #15180: Errors with "make px4_sitl gazebo" --> recipe for target 'px4_sitl' failed.

MKSwaminathan, 2020-06-21:
Hello!

Here are my stats:
Ubuntu version: 18.04
Gazebo version: 9.13.1
PX release: v1.9.0
ROS melodic

When I try and run:
make clean
make px4_sitl gazebo

I get the following error:
[error.log](https://github.com/PX4/Firmware/files/4810258/error.log)

Please let me know how I can fix this/what I may be doing wrong. Thank you for your support!

Jaeyoung-Lim, 2020-06-23:
@MKSwaminathan Could you maybe try the latest master and see if the issue is still there? I don't think PX4 v1.9.0 was ever tested against gazebo version 9.13.1

gelardrc, 2020-06-23:
> Hello!
>
> Here are my stats:
> Ubuntu version: 18.04
> Gazebo version: 9.13.1
> PX release: v1.9.0
> ROS melodic
>
> When I try and run:
> make clean
> make px4_sitl gazebo
>
> I get the following error: [...]

I try this and work fine for me.

sudo apt-get install libprotobuf-dev libprotoc-dev protobuf-compiler libeigen3-dev libxml2-utils python-rospkg python-jinja2

sudo apt-get install libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly -y

Maybe you'll need to install more lib than that.

MKSwaminathan, 2020-06-23:
Thank you both for all your time and help!
@Jaeyoung-Lim I got the same issue building the latest master, but I think this was a machine specific issue (although I'm curious as to why).
@gelardrc I tried installing those and realized that already have those dependencies installed, but here's how I finally fixed the issue: 
When I ran a standalone build of the Firmware/Tools/sitl_gazebo, the error log revealed that it was a linker issue, and it was trying to link dynamic and static libraries. [Here is an explanation of why this is the case.](https://stackoverflow.com/questions/54423834/why-cant-you-statically-link-dynamic-libraries) I globally appended the -fPIC flag

`export CXXFLAGS='$CXXFLAGS -fPIC' ` 
and then 
`./configure --enable-shared` 

I then rebuilt px4 by making a build dir, running cmake etc. Somehow that fixed everything.

closed 2020-06-23 by MKSwaminathan