PX4/PX4-Autopilot #15184: IMU_GYRO_RATEMAX in SITL

Sunchanghao, 2020-06-22:
The questions are as follows:
I need modify the parameters in SITL world file like that:
    ```
 <max_step_size>0.001</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>1000</real_time_update_rate>
```
change max_step_size from 0.004 to 0.001, and then change eal_time_update_rate from 250 to 1000.
 In order to work properly it had to modify the  IMU_GYRO_RATEMAX from 400 to 1000.
after that, iris can fly but it's not working properly, it is hard to takeoff and landing, meanwhile,　the flight mode also  work bad.
such as the px4 is under HOLD mode, when I push any joystick, px4 will switch mode to Position mode, and when I control throttle to 0(iris is on the ground), the rotors are not stop, and from QGC I can find that the altitude is inaccurate(high volatility).

bys1123, 2020-06-22:
@Jaeyoung-Lim @TSC21 Do you hava any idea on this, thanks!

dagar, 2020-06-22:
At the moment with lockstep it's effectively required that the estimator runs at the IMU rate. That means the simulator rate, `IMU_GYRO_RATEMAX`, and `IMU_INTEG_RATE` should all be set the same.

Sunchanghao, 2020-06-23:
@dagar 
Thanks
I try to change IMU_INTEG_RATE from 250 to 1000, and but nothing has changed.
 so if  I need to set <max_step_size>0.001</max_step_size> ,first  I change IMU_INTEG_RATE from 250 to 1000,and for IMU_GYRO_RATEMAX,  I dont know why it is 400hz and I change it to 50/1000/2000, but the sensor is still not working properly.

wangwwno1, 2022-03-11:
Hello, I also got similar problem here.
I'm using jMAVSim SITL, and changing the fresh rate in `sitl_run.sh` from 250 to 400.
But the PX4 said `poll timeout 0, 22` and failed to start up.

So here is my question: is it possible to run the SITL at a freshing rate faster than 250Hz?