PX4/PX4-Autopilot #15201: Pixhawk 4 can't connect to DSMX receiver

edwinpha, 2020-06-23:
I am using a DSMX 9645 receiver with a radio DX8. 

PX4 does not recognize any signal given by the receiver. I have tried using spektrum bind and nothing happens. The receiver's led does not even begin blinking.

I tried the same components on Betaflight (another board) and Ardupilot, and there are no problems in binding or reading transmitted signals.

I have even tried binding it with another autopilot and then trying it on PX4. From the receiver's led, I can notice that it is bind with the radio. However, it still does not recognize any signal on any channel.

dagar, 2020-06-23:
Did you try Ardupilot on the same Pixhawk 4 or a different board?

edwinpha, 2020-06-23:
Same board

taileron, 2020-06-24:
It seems to be a very similar situation like for me on Pixfalcon.
If a in 1.10 successful binded DSMX receiver appears one time after the first boot after updating from 1.10 to 1.11 it would be the same behaviour and only something in the initialization procedure timing has to be corrected. 
It you have a voltmeter, you can check if and when the 3.3V appear on the red wire to the satellite.
The binding created in other versions is preserved in any case and does not need to be rebound for 1.11

taileron, 2020-06-26:
sys_use_io = 0 enables the rc function but disables the whole io-board so rc values come only from fmu

nrogelio, 2020-08-20:
Hi,

A PR was merged to master: https://github.com/PX4/Firmware/pull/15430

You can try to write master firmware to your board and see if that solves the issue. Of course master firmware is a developer version so use it at your own risk.

Regards.

edwinpha, 2020-10-19:
Thanks for the message @nrogelio, I have been a little busy so I have tried only now.

The situation is the same, the radio is bound with the receiver but PX4 says that I have zero channels.
Today, I even tried DSMX 9645 with AR8010T but the results are always the same.

I have read #15430, but I am not sure if @dagar could emulate this exact problem and solve it.

MrXermon, 2020-10-29:
I am having the same issue, currently. I verified that the receiver is working with Ardupilot but I cannot get it running with this firmware. I've checked the stable and the master release without any success.

taileron, 2020-10-30:
my problem is practically solved with the parameter sys_use_io = 0 you can try if it is the same mixer problem, if so this mixer can be operated with 6 channels on the FMU without IO. Loading a custom mixer with more than 2 channel entries to the IO blocks the spectrum receiver -> manual control lost if the rc setup has been completed successfully or if a new adjustment is made 0 rc channels are detected.  The rc blocking mixer can be loaded successfully later with an entry into the sd cards etc/extras.txt 
mixer load /dev/pwm_output0 /fs/microsd/etc/mixers/vtol_test.mix

ilNonn0, 2020-12-05:
I am facing the same problem trying to get my Spektrum DX8 G2 working with PX4 firmware. 
By using QGC I am not able to bind the Spektrum receiver ([remote satellite SPM9645](https://www.spektrumrc.com/Products/Default.aspx?ProdId=SPM9645)).
Therefore I manually bound the receiver with the transmitter and then I attached it to pixhawk4 (on DSM/SBUS RC port), but the QGC is not able to see the receiver and pressing the _Calibrate_ button on the Radio setup page, I got the following error message: "detected 0 radio channels. to operate px4 you need at least 5 channels".
During the procedure, the orange LED on the receiver is always on (solid orange) meaning that the transmitter and the receiver are correctly bound and are communicating. 

As a proof, to understand if the cables were ok, the wiring was fine etc., I flashed Ardupilot firmware without touching anything. I found that with such a firmware the system works. I such a case I can see the radio channels, to calibrate the radio and to bind the receiver (again).

How can I solve the problem?
Are there some parameters that I have to set up and that I am missing?

@dagar the PX4 firmware version I tested was the last stable release (v1.11.0) and the master one (last build on 4 December).

@taileron I tried changing the parameter as you suggested, but it didn't work

Thanks for your support

LorenzMeier, 2020-12-11:
I'm working on this in this issue: https://github.com/PX4/PX4-Autopilot/issues/16358 - I will close this one as duplicate.

closed 2020-12-11 by LorenzMeier