PX4/PX4-Autopilot #15205: PX4 cube mission planner setting (using xbee  S2C pro module )

sinanoksuz, 2020-06-23:
ı  use px4 cube , ı use telemetry as two Xbee S2C pro  (router coordinator)  But there is some issue , this is my setting on XCTU programs  
[image]

[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]

Can you Help me . What kind of issue in this situation  ?  there is issue on missionplanner program)

julianoes, 2020-06-24:
This GitHub repository is for PX4. Your issue presumably has to do with MissionPlanner which works with ArduPilot (and not PX4).

I would ask your question in: https://github.com/ArduPilot/MissionPlanner

closed 2020-06-24 by julianoes

sinanoksuz, 2020-06-24:
okey thank you