PX4/PX4-Autopilot #15211: Differential rover goes into a spin when GPS position is lost

Jaeyoung-Lim, 2020-06-24:
**Describe the bug**
When GPS is lost, differential rover goes into a fixed bank loiter - which results in the vehicle spinning around while being stationary. 

**To Reproduce**
Steps to reproduce the behavior:
1. `make px4_sitl gazebo_r1_rover`
2. Start a mission
3. Trigger a gps failure (e.g. gps stop in mavlink shell)

**Expected behavior**
Not sure, but maybe keeping stationary would be a more sensible behavior

**Log Files and Screenshots**
[image]

dagar, 2020-06-26:
Quick fixes that come to mind.
1. Hack navigator/gpsfailure for rover     
     - https://github.com/PX4/Firmware/blob/1761ef39520a271982a7a01ef6cd2b934d359a1b/src/modules/navigator/gpsfailure.cpp#L86-L90
2. Handle it in the state machine.
    - https://github.com/PX4/Firmware/blob/1761ef39520a271982a7a01ef6cd2b934d359a1b/src/modules/commander/state_machine_helper.cpp#L701-L759
    - A special case here for rover would be tolerable. https://github.com/PX4/Firmware/blob/1761ef39520a271982a7a01ef6cd2b934d359a1b/src/modules/commander/state_machine_helper.cpp#L730-L745

Jaeyoung-Lim, 2020-06-26:
Realized this was a duplicate of https://github.com/PX4/Firmware/issues/13467

closed 2020-11-28 by Jaeyoung-Lim

aaryanmurgunde, 2022-06-02:
@Jaeyoung-Lim Was a solution found for this ?

Jaeyoung-Lim, 2022-06-02:
@aaryanmurgunde No, as you can see in the thread https://github.com/PX4/PX4-Autopilot/issues/13467 is still open

aaryanmurgunde, 2022-06-02:
I was trying to do a quick simulation on any differential drive robot with either Rover by Ardupilot or PX4 Autopilot 
Do you have any alternative that I could use for SITL and Gazebo ROS?

@Jaeyoung-Lim

Jaeyoung-Lim, 2022-06-02:
@aaryanmurgunde This issue is not related to the model being usable or not but related to the behavior of what happens when GPS is lost. You can use the model as is

aaryanmurgunde, 2022-06-02:
@Jaeyoung-Lim So if I get this right I'm using SITL and gazebo so technically the GPS should be mimicked by the SITL right ?

aaryanmurgunde, 2022-06-02:
@Jaeyoung-Lim In my case I don't see a gps failure still face the same issue 
[image]

aaryanmurgunde, 2022-06-02:
You can see all three screens here 
https://user-images.githubusercontent.com/44398228/171621239-ff323636-a778-4751-937e-596f29484458.mp4

sajisaka, 2023-03-21:
Any update on this issue?
I'm in the same situation, but it doesn't seem to be a GPS problem.
When the rover starts to spin, the GPS is still capturing the position.

Jaeyoung-Lim, 2023-03-21:
@sajisaka If that is the case, it is not the same problem