PX4/PX4-Autopilot #15217: Suggestion - moving EXTENDED_SYS_STATE data to HEARTBEAT:system_status

BazookaJoe1900, 2020-06-24:
there are only two data fields on _EXTENDED_SYS_STATE_: _vtol_state_ and _landed_state_. and each one are 1 byte.
both can be moved to _HEARTBEAT:system_status_ field which is 4 byte field, and two of them are not even used and saved as reserved.

there are few logic to do such change:
- it looks like waste of bandwidth
- system_status is exactly the place to put the flight mode of the drone (_vtol_state_ and _landed_state_)

there is at least one reason NOT to do the change:
- removing the use of EXTENDED_SYS_STATE is major change, and critical. because QGC then allows to take off, but because it has no "landed_state" state, it doesn't shows the land button.

any thoughts?

julianoes, 2020-06-25:
> HEARTBEAT:system_status field which is 4 byte field

No, that's just a `uint8` :thinking: 

I believe such a change would have to be specced on the mavlink level first so with an issue in https://github.com/mavlink/mavlink. However, I'm not very optimistic that it would be easy to change the meaning of anything in heartbeat because that message is used everywhere.

BazookaJoe1900, 2020-06-25:
@julianoes, my mistake about the field name, I meant custom_mode

julianoes, 2020-06-25:
Aha, that's an idea. So it would be in these two bytes: https://github.com/PX4/Firmware/blob/master/src/modules/commander/px4_custom_mode.h#L76

I'm not convinced the effort to change this everywhere is worth the optimization that we gain from it. Given we probably still need to send EXTENDED_SYS_STATE for a while until ground stations have adapted.

BazookaJoe1900, 2020-07-17:
@julianoes @dagar 
I understand the suggestion doesn't sound good for you? should I close the issue?

dagar, 2020-07-17:
I think it's too hard to change the existing HEARTBEAT. I still dream of the day we can get beyond the custom_mode mess.

BazookaJoe1900, 2020-07-17:
its not too hard to change the HEARTBEAT , but it breaks compatibility.
it can be done in steps, that on the first step don't remove EXTENDED_SYS_STATE message, and the data will be redundant on HEARTBEAT and EXTENDED_SYS_STATE.
Do you want me to do pr about that? (its abit of work, that I don't want to do without some approval that it **might** be used)

dagar, 2020-07-17:
Yes when I say it's too hard I'm entirely referring to the compatibility headache. Some of it we can control and gradually change, but quite a lot of it we can't.

If the goal is optimization I think there's quite a bit more we could do before getting into compatibility issues. Quite a lot of the data sent by default isn't needed most of the time. Messages that are both larger and at higher rates than `EXTENDED_SYS_STATE`.

For example `RC_CHANNELS` is much larger than `EXTENDED_SYS_STATE` and sent by default at 5x the rate. This message is only ever used for RC calibration, and hence it's a waste in virtually all normal operation. Changing it to only stream when needed during RC calibration would be a much bigger savings without the compatibility difficulties.

There are several relatively easy mavlink usage optimization cases like this. There's also quite a lot of potential improvement in the mavlink module itself properly tracking and throttling the data rate and some kind of flow control at the mavlink level. I'm happy to go into more detail if you're interested.

closed 2020-07-17 by BazookaJoe1900