PX4/PX4-Autopilot #15223: Stuck baro leads to flyaway

julianoes, 2020-06-26:
**Describe the bug**
As part of the failsafe testing in #15063 I tried out what happens when the baro value is stuck, so the identical value keeps getting published. I could see that this could happen when something physically blocks the baro.

What seems to happen in this case is that altitude is controlled against stale baro data which can lead to a flyaway.

**To Reproduce**
Check out #15063 with https://github.com/mavlink/MAVSDK/pull/1124.
And run:
```
DONT_RUN=1 make px4_sitl gazebo mavsdk_tests && test/mavsdk_tests/mavsdk_test_runner.py test/mavsdk_tests/configs/sitl.json --abort-early --speed-factor 1 --iterations 1 --case 'Continue on baro stuck during mission' --gui --verbose
```

**Expected behavior**
I would expect the ekf2 to stop rejecting baro altitude and switch to GPS height.

**Log Files and Screenshots**

https://logs.px4.io/plot_app?log=e835416d-c1cd-47c8-8cc0-e9142e71b209

dagar, 2020-06-26:
Looks like a bug in sensors (at least first).

[image]

julianoes, 2020-07-10:
Fixed by #15273.

closed 2020-07-10 by julianoes