PX4/PX4-Autopilot #15225: MacOS intermittent segmentation fault during unit tests

dagar, 2020-06-26:
https://github.com/PX4/Firmware/pull/15222/checks?check_run_id=810620037

[image]

Only seems to be happening on the Github Actions MacOS machines, and not the older Jenkins MacOS build slaves.

dagar, 2020-06-26:
@julianoes we should think about replicating the core dump, backtrace, and upload functionality from the Linux SITL tests.

dagar, 2020-06-26:
Github Actions MacOS (currently 10.15) environment details. https://github.com/actions/virtual-environments/blob/master/images/macos/macos-10.15-Readme.md

dagar, 2020-06-29:
Running the tests on Github Actions (MacOS) has been disabled for now, but this will still need to be addressed.

Jaeyoung-Lim, 2020-12-03:
@dagar Should we bring this back? macOS builds properly with gazebo now

dagar, 2020-12-03:
> @dagar Should we bring this back? macOS builds properly with gazebo now

Yes let's give it a try.

closed 2020-12-18 by Jaeyoung-Lim