PX4/PX4-Autopilot #15226: Gazebo SITL is not working with Jetson

netgittemp, 2020-06-26:
Hi, I am working with Jetson and SITL.

I installed everything and I runed
`~/src/Firmware$ sudo make px4_sitl_default gazebo`
But I cannot see any px4 fcu connected.
Because when I hit the enter key and I cannot see any prompt for "pxh>" in order to put commands.
This is resulting that I cannot run offboard mode package test neither.

The installation instruction is working very well with my laptop, but not with jetson.
Please help me to resolve this issue.
Thank you in advance.

Jaeyoung-Lim, 2020-06-27:
@netgittemp Please provide more information such as the actual log of the failure. Otherwise it is hard to determine what is not working / working

netgittemp, 2020-06-28:
The different between my laptop output and the jetson output is
```

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.61
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2020-06-26/17_44_06.ulg
INFO  [logger] Opened full log file: ./log/2020-06-26/17_44_06.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> [Wrn] [GuiIface.cc:120] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'

PX4 Exiting...
pxh> Exiting NOW.
ninja: build stopped: interrupted by user.
Makefile:217: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Interrupt
```

I hope this is enough to understand the issue.
If not, please let me know, then I will provide everything you want!

netgittemp, 2020-06-30:
@Jaeyoung-Lim 
Is there any idea about this?
It is very weird that the same way is not working with Jetson...
I reflashed ubuntu 18.04 and reinstall everything step by step, but it is not working.

netgittemp, 2020-07-01:
This is what I got, and no pxh>

[image]

netgittemp, 2020-07-01:
@Jaeyoung-Lim 
Could you please help me to make it accept the TCP connection?

Thank you.

Jaeyoung-Lim, 2020-07-01:
@netgittemp First of all, the jetson uses a aarch64 architecture and not a amd64 processor as normal desktop / laptops. Since we do not test SITL against arm based systems it is not surprising that it doesn't work. 

1. Could you try to run just gazebo and see if you can run it?
2. Can you run SITL in headless mode? (without gazebo gui
3. Have you tried other simulators such as jmavsim ?

netgittemp, 2020-07-02:
Thank you for your reply.

1. Gazebo is working correctly, even with `sudo make px4_sitl_default gazebo'.
2. What will be the command for it?
Is `HEADLESS=1 make px4_sitl_default iris` fine?
3. I did run the simulation with jmavsim. I will do it soon tomorrow.

It would be very thankful if you can comment regarding "TCP port connection".

Thank you in advance.

Jaeyoung-Lim, 2020-07-05:
@netgittemp 
1. why are you running the command with sudo? something is wrong

2. Okay, so you are unable to run px4

3. any updates?

netgittemp, 2020-07-11:
@Jaeyoung-Lim 
I am sorry that my tardy response.

1. Because I got the following error when I run `$ make px4_sitl_default gazebo` (without `sudo`)
```
ninja: error: opening build log: Permission denied
Makefile:217: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1
```

2. In regard to jmavsim, the output of `$ sudo make px4_sitl_default jmavsim` is
it is the same output of TCP issue.

I am wondering if there is someone else who was troubling with the same TCP issue with me.

julianoes, 2020-07-14:
If you have used `sudo` before you first need to fix the permissions again.
```
cd Firmware
chown -R $USER:$USER .
make clean
make px4_sitl jmavsim
```

Jaeyoung-Lim, 2021-01-13:
Closing as original issuer did not respond. Please re-open if this is still an issue

closed 2021-01-13 by Jaeyoung-Lim

JohnnyLiu2886, 2021-11-15:
I met the same problem. I am using jetson nano with ubuntu 18.04. I found the problem is that, the simulation is stuck in "INFO [simulation] waiting for simulator to accept connection on TCP port 4560" 

I have no idea why it is happening on jetson. I have tested them on both jetson nano and jetson xavier. They comes out with the same result. Does anyone using ubuntu 18.04 can run "make px4_sitl_rtps gazebo" successfully? The problem is absolutely not from gazebo. It is because when I ran it with "HEADLESS=1 make px4_sitl_rtps gazebo", it is still stuck in "INFO [simulation] waiting for simulator to accept connection on TCP port 4560". 

Here is my result:
[image]

Here is the successful case from internet:
[image]