PX4/PX4-Autopilot #15238: mavlink_sitl_gazebo fails in catkin build (gtest)

pariaspe, 2020-06-29:
I had the PX4-ROS-gazebo infrastructure working well, but when I rebuilt it yesterday, it stopped working. The problem seems to be with GTEST libraries.

I just did a `catkin clean` and a `catkin build`. This is what I got:

```bash
-------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic:/opt/jderobot:/opt/jderobot
Workspace:                   /home/parias/catkin_ws
-------------------------------------------------------------------------
Build Space:        [exists] /home/parias/catkin_ws/build
Devel Space:        [exists] /home/parias/catkin_ws/devel
Install Space:      [unused] /home/parias/catkin_ws/install
Log Space:         [missing] /home/parias/catkin_ws/logs
Source Space:       [exists] /home/parias/catkin_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------------------
[build] Found '11' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 2.0 seconds ]             
Starting  >>> mavlink                                                          
Starting  >>> mavros_auto_mission                                              
Starting  >>> mavros_msgs                                                      
Finished  <<< mavros_auto_mission                  [ 3.2 seconds ]             
Finished  <<< mavlink                              [ 6.0 seconds ]             
Starting  >>> libmavconn                                                       
Finished  <<< mavros_msgs                          [ 13.4 seconds ]            
Finished  <<< libmavconn                           [ 6.6 seconds ]             
Starting  >>> mavros                                                           
Starting  >>> px4                                                              
Finished  <<< mavros                               [ 1 minute and 23.6 seconds ]
Starting  >>> drone_wrapper                                                    
Finished  <<< drone_wrapper                        [ 5.5 seconds ]             
Starting  >>> mavlink_sitl_gazebo                                              
Starting  >>> mavros_extras                                                    
Starting  >>> rqt_drone_teleop                                                 
_______________________________________________________________________________
Errors     << mavlink_sitl_gazebo:cmake /home/parias/catkin_ws/logs/mavlink_sitl_gazebo/build.cmake.000.log
CMake Error at /opt/ros/melodic/share/catkin/cmake/assert.cmake:3 (message):

  Assertion failed: GTEST_LIBRARIES (value is '')

Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:185 (assert)
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:89 (_catkin_add_executable_with_google_test)
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:37 (_catkin_add_google_test)
  unit_tests/CMakeLists.txt:17 (catkin_add_gtest)
  unit_tests/CMakeLists.txt:31 (add_unit_test)

cd /home/parias/catkin_ws/build/mavlink_sitl_gazebo; catkin build --get-env mavlink_sitl_gazebo | catkin env -si  /usr/bin/cmake /home/parias/catkin_ws/src/mavlink_sitl_gazebo --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/parias/catkin_ws/devel/.private/mavlink_sitl_gazebo -DCMAKE_INSTALL_PREFIX=/home/parias/catkin_ws/install; cd -
...............................................................................
Failed     << mavlink_sitl_gazebo:cmake            [ Exited with code 1 ]      
Failed    <<< mavlink_sitl_gazebo                  [ 9.7 seconds ]             
Abandoned <<< test_mavros                          [ Unrelated job failed ]    
Finished  <<< rqt_drone_teleop                     [ 3.7 seconds ]             
Finished  <<< px4                                  [ 1 minute and 41.0 seconds ]
Finished  <<< mavros_extras                        [ 41.8 seconds ]            
[build] Summary: 10 of 12 packages succeeded.                                  
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2 minutes and 39.2 seconds total.                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```
I use ROS Melodic, Gazebo 9.13.1 and Ubuntu 18.04.06. Also, I cloned _Firmware_ from **master** and  from **stable** branch, but I get the same error. 

Thanks!

adbidwai, 2020-07-03:
Make sure your submodules are upto date. Update them using ``git submodule update --recursive --init``

Jaeyoung-Lim, 2020-07-05:
@pariaspe  `sitl_gazebo` should not be built as part of a catkin package. If you want to use it in ROS, follow the build instructions and configure the path properly as written in https://dev.px4.io/master/en/simulation/ros_interface.html

closed 2020-07-05 by Jaeyoung-Lim