PX4/PX4-Autopilot #15245: it's report "error: [Errno 104] Connection reset by peer",when  running sim_vehicle.py --console --map

theonly-unicorn, 2020-06-30:
it's report "error: [Errno 104] Connection reset by peer",when  running sim_vehicle.py --console --map

PS:i use SITL simulator

theonly-unicorn, 2020-06-30:
Loaded module map
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_map/mp_tile.py", line 276, in downloader
    resp = url_open(req)
  File "/usr/lib/python2.7/urllib2.py", line 127, in urlopen
    return _opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 404, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 422, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 382, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1214, in http_open
    return self.do_open(httplib.HTTPConnection, req)
  File "/usr/lib/python2.7/urllib2.py", line 1187, in do_open
    r = h.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1089, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 444, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 400, in _read_status
    line = self.fp.readline(_MAXLINE + 1)
  File "/usr/lib/python2.7/socket.py", line 476, in readline
    data = self._sock.recv(self._rbufsize)
error: [Errno 104] Connection reset by peer

theonly-unicorn, 2020-06-30:
What is the cause of this and how to solve it

julianoes, 2020-07-10:
It looks like you are doing something with MAVProxy. I don't think that has anything to do with PX4. I would ask in https://github.com/ArduPilot/MAVProxy.

closed 2020-07-10 by julianoes

Jaeyoung-Lim, 2020-07-10:
`sim_vehicle.py` is a script to start SITL in Ardupilot, not PX4