PX4/PX4-Autopilot #15253: Comment fix

George-avy, 2020-07-03:
https://github.com/PX4/Firmware/blob/400c44b5ef692838aa6f6ac040eb0dff7536b2e8/src/lib/rc/sbus.h#L64

Actually at 100kbps, 25bytes per frame and 8E2 (12bits per character) UART configuration , frame duration is exactly 3ms:

     25chars * 12bits/char / 100000bits/s = 3ms

George-avy, 2020-07-03:
Same here: https://github.com/AvyFly/PXAvy/blob/stable_candidate/src/lib/rc/sbus.cpp#L306

julianoes, 2020-07-10:
Thanks for the note. We would appreciate a pull request with your suggested changes, thanks.

George-avy, 2020-07-10:
> Thanks for the note. We would appreciate a pull request with your suggested changes, thanks.
Yeah, I know...

nrogelio, 2020-08-24:
Hi George-avy,

I hope you don't mind I created a PR to add your changes.

Best Regards.

closed 2020-08-24 by bresch

George-avy, 2020-08-24:
> Hi George-avy,
> 
> I hope you don't mind I created a PR to add your changes.
> 
> Best Regards.

Why would I mind? Good job!

PR #15605 (2020-08-24): Sbus comment fix, found by George-avy in issue 15253
Fix to issue #15253 found by George-avy

fixes #15253