PX4/PX4-Autopilot #15265: 6 motors in dodecacopter configuration not working

neom89, 2020-07-06:
I have recently started testing on Dodecacopter airframe. For this, I have connected 6 motors in main port of Pixhawk 2 and the other 6 motors on the AUX port.

After this on arming the setup, Only 6 motors on the main port are rotating and the other 6 which are connected to the AUX port are still beeping.

Airframe: Dodecacopter coaxial
sys_autostart: 24001
Firmware: Daily build

julianoes, 2020-07-10:
You can try to debug what is going on by doing `pwm info -d /dev/pwm_output0` for MAIN pins and `pwm info -d /dev/pwm_output1` for AUX pins in a system console or mavlink shell.

neom89, 2020-07-12:
@dagar @julianoes Here is the out from pwm info command for both main and aux pins:

[image]

I tried loading the mixer file for aux outputs manually but got the following error:

[image]

dagar, 2020-07-13:
That's surprising, it seems to be a problem fmu side rather than the extremely memory limited px4io. Have you tried with current master (or beta)? The `fmu` module is now called `pwm_out`.

neom89, 2020-07-13:
@dagar I have tried with both Stable release v1.10.1 and beta.

neom89, 2020-07-13:
@dagar I just checked with current master and I am getting the same outputs/errors.

CornerOfSkyline, 2020-07-14:
> @dagar @julianoes Here is the out from pwm info command for both main and aux pins:
> 
> [image]
> 
> I tried loading the mixer file for aux outputs manually but got the following error:
> 
> [image]

Manual to load mixer, the command squence is "mixer load 'device' 'mix file' "
It should be :
`mixer load /dev/pwm_output1 /etc/mixers/dodeca_bottom_cox.aux.mix`
@neom89

neom89, 2020-07-14:
@CornerOfSkyline @dagar @julianoes 
I used the wrong syntax int the last post as suggested by @CornerOfSkyline. When I use the correct command, the mixer for dodeca still can not be loaded manually. Here is the output of the command:

[image]

However, if I try to load another mixer from /etc/mixer, the mixer is loaded successfully. Here is the output of the command:

[image]