PX4/PX4-Autopilot #15268: error: ‘mavlink_distance_sensor_t {aka struct __mavlink_distance_sensor_t}’ has no member named ‘signal_quality’

Rezenders, 2020-07-06:
**Describe the bug**
I am trying to build this repo inside a catkin workspace with catkin_make and I am getting the following error:

```
[ 31%] Building CXX object sitl_gazebo/CMakeFiles/gazebo_magnetometer_plugin.dir/src/geo_mag_declination.cpp.o
/home/gus/Documents/git/jason_ros_ws/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::LidarCallback(gazebo::LidarPtr&, const int&)’:
/home/gus/Documents/git/jason_ros_ws/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:1048:14: error: ‘mavlink_distance_sensor_t {aka struct __mavlink_distance_sensor_t}’ has no member named ‘signal_quality’
   sensor_msg.signal_quality = lidar_message->signal_quality();
              ^~~~~~~~~~~~~~
/home/gus/Documents/git/jason_ros_ws/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::SonarCallback(gazebo::SonarPtr&, const int&)’:
/home/gus/Documents/git/jason_ros_ws/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:1120:14: error: ‘mavlink_distance_sensor_t {aka struct __mavlink_distance_sensor_t}’ has no member named ‘signal_quality’
   sensor_msg.signal_quality = sonar_message->signal_quality();
              ^~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-address-of-packed-member’
sitl_gazebo/CMakeFiles/gazebo_mavlink_interface.dir/build.make:62: recipe for target 'sitl_gazebo/CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o' failed
make[2]: *** [sitl_gazebo/CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o] Error 1
CMakeFiles/Makefile2:12014: recipe for target 'sitl_gazebo/CMakeFiles/gazebo_mavlink_interface.dir/all' failed
make[1]: *** [sitl_gazebo/CMakeFiles/gazebo_mavlink_interface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```
**Additional context**
I am using the master version (96a6b5c)

hamishwillee, 2020-07-06:
@Rezenders signal_quality is a new member of distance_sensor. Have you updated submodules  or cleaned your PX4 build? I.e. `git submodule update --recursive` or `make distclean`?

TSC21, 2020-07-06:
> **Describe the bug**
> I am trying to build this repo inside a catkin workspace with catkin_make and I am getting the following error:
>
> ```
> [ 31%] Building CXX object [...]

You will keep having this issue since the ROS Mavlink release needs to be up to date with the Mavlink current generated headers, which is not the case. This happens because since you are building inside a catkin_ws, the first headers to be retrieved are the ones available in the ROS install dir.

Nothing that can be done until a new ROS Mavlink release is out, which we do not control. @vooon FYI.

closed 2020-07-06 by TSC21