PX4/PX4-Autopilot #15274: Not returning home when Failsafe mode is set

Rezenders, 2020-07-07:
**Describe the bug**
Hello,

I am running a gazebo simulation using mavros to fly trough some setpoints and then land. Everything was working just fine until I updated ubuntu. I have a program that is sending some local setpoints for a while then it stops. Before, when I stoped sending setpoints the UAV would enter the `Failsafe mode` and return to home. Now, when the UAV enters the `Failsafe mode` it holds its position and stays in `"AUTO.LOITER"` mode.

I checked the [COM_OBL_ACT](https://docs.px4.io/v1.9.0/en/advanced_config/parameter_reference.html#mission) parameter and it is set to 2 which means it should return to home. I do not know why it is not returning home.

Btw, I get this message when Failsafe mode is set:

    [ERROR] [1594071660.258264937, 24.964000000]: FCU: Failsafe enabled: no RC and no offboard
    [ INFO] [1594071660.267359507, 24.972000000]: WP: reached #0
    [ INFO] [1594071660.310152532, 25.012000000]: FCU: Failsafe mode activated

PS:
I updated the px4 Firmaware to the current master [96a6b5c](https://github.com/PX4/Firmware/commit/96a6b5c914afa44a01aa782944dce034ce6b4a2b)

Setting the UAV mode to` "AUTO.LAND"` does not work, it stays in `"AUTO.LOITER"`

**Log Files and Screenshots**
- [Logfile](https://review.px4.io/plot_app?log=ae3246bd-edee-4159-87ba-855da3f34211)

Rezenders, 2020-07-30:
Sorry to bother. Any updates on this? Has anyone else had this same problem? Any chance that I'm doing something wrong?

julianoes, 2020-07-31:
Thanks for the bump. I'll try to cover that in #15063 in the next weeks.