PX4/PX4-Autopilot #15275: v1.11.0-rc1 build fails (-Werror=unused-but-set-parameter)

Roman-, 2020-07-07:
Trying to build v1.11.0-rc1 on Ubuntu 18.04.2 LTS.

`make px4-sitl gazebo` builds without any errors.

However, `make px4_fmu-v4_default` gives me this:

<pre>
In file included from /home/user/repos/px4/Firmware-official/platforms/common/include/px4_platform_common/atomic_bitset.h:36:0,
                 from /home/user/repos/px4/Firmware-official/src/modules/uORB/uORBDeviceMaster.hpp:54,
                 from /home/user/repos/px4/Firmware-official/src/modules/uORB/uORBDeviceNode.hpp:37,
                 from /home/user/repos/px4/Firmware-official/src/modules/uORB/Publication.hpp:45,
                 from /home/user/repos/px4/Firmware-official/src/modules/uORB/PublicationMulti.hpp:45,
                 from /home/user/repos/px4/Firmware-official/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:40,
                 from /home/user/repos/px4/Firmware-official/src/drivers/imu/invensense/icm20608g/ICM20608G.hpp:46,
                 from /home/user/repos/px4/Firmware-official/src/drivers/imu/invensense/icm20608g/ICM20608G.cpp:34:
/home/user/repos/px4/Firmware-official/platforms/common/include/px4_platform_common/atomic.h: In instantiation of 'bool px4::atomic<T>::compare_exchange(T*, T) [with T = unsigned char]':
/home/user/repos/px4/Firmware-official/src/drivers/imu/invensense/icm20608g/ICM20608G.cpp:385:109:   required from here
/home/user/repos/px4/Firmware-official/platforms/common/include/px4_platform_common/atomic.h:169:34: error: parameter 'expected' set but not used [-Werror=unused-but-set-parameter]
  inline bool compare_exchange(T *expected, T desired)
                                  ^~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
src/drivers/imu/invensense/icm20608g/CMakeFiles/drivers__imu__invensense__icm20608g.dir/build.make:62: recipe for target 'src/drivers/imu/invensense/icm20608g/CMakeFiles/drivers__imu__invensense__icm20608g.dir/ICM20608G.cpp.obj' failed
make[4]: *** [src/drivers/imu/invensense/icm20608g/CMakeFiles/drivers__imu__invensense__icm20608g.dir/ICM20608G.cpp.obj] Error 1
make[4]: *** Waiting for unfinished jobs....
</pre>

julianoes, 2020-07-10:
What's your `arm-none-eabi-gcc --version`?

Roman-, 2020-07-10:
> What's your `arm-none-eabi-gcc --version`?

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437], got it from here: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads/6-2017-q2-update.

Ubuntu 16.04LTS with the same tool produces the same result.

xdwgood, 2020-07-14:
I encountered the same problem using the master branch

closed 2020-07-14 by dagar