PX4/PX4-Autopilot #15280: SITL vehicle does not spawn at 0,0,0 in Gazebo

vorobotics, 2020-07-08:
In [sitl_run.sh](https://github.com/PX4/Firmware/blob/38fa913fedf49b65d24502182d5af601d41e891b/Tools/sitl_run.sh#L108), line 108, is there a reason why the model spawns at (1.01, 0.98, 0.83) instead of exactly at (0,0,0)? 

Here's the line I'm referring to: 
`gz model --spawn-file="${src_path}/Tools/sitl_gazebo/models/${model}/${model}.sdf" --model-name=${model} -x 1.01 -y 0.98 -z 0.83`

Jaeyoung-Lim, 2020-07-08:
@vorobotics Nothing major, but the intention was too not spawn it at 0, 0, 0 since this is a special case, and sometimes hides issues in your code

vorobotics, 2020-07-08:
OK, that makes sense. Perhaps the [documentation](https://dev.px4.io/master/en/simulation/gazebo.html) should be updated, as it currently says:
> The vehicle gets spawned into the origin of the world model

Jaeyoung-Lim, 2020-07-08:
@vorobotics Would you be able to add it to the documentation through https://github.com/PX4/Devguide ?

hamishwillee, 2020-07-08:
How about adding a note for this ... i.e. 

The vehicle gets spawned into the origin of the world model at some simulated GPS location.

> **Note** The vehicle is actually spawned *close* to the origin (0,0,0), but not exactly on it.

closed 2020-07-08 by vorobotics