PX4/PX4-Autopilot #15282: Creating a Fast RTPS Listener application and run,but nothing happened

Yoke365, 2020-07-08:
I follow the step in https://dev.px4.io/master/en/middleware/micrortps.html
## Setp1 
[Client (PX4 Firmware)](https://dev.px4.io/master/en/middleware/micrortps.html#client_firmware)
```
make px4_sitl_rtps jmavsim
```
The Client application can be launched from System Console. 
[image]

## Step2
The step in [Agent in a ROS-independent Offboard Fast RTPS interface] (https://dev.px4.io/master/en/middleware/micrortps.html#agent-in-a-ros-independent-offboard-fast-rtps-interface) for other terminal
```
cd build/<px4_sitl_rtps>/src/modules/micrortps_bridge/micrortps_client/micrortps_agent
mkdir build && cd build
cmake ..
make
```
run Agent application
[image]

## Setp3
The step in [Creating a Fast RTPS Listener application](https://dev.px4.io/master/en/middleware/micrortps.html#creating-a-fast-rtps-listener-application)
- Enter the following commands to create the application:
```
cd /path/to/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge
mkdir micrortps_listener
cd micrortps_listener
fastrtpsgen -example x64Linux2.6gcc ../micrortps_client/micrortps_agent/idl/sensor_combined_.idl
```
ps:  The last sentence of the document is missing an underline, but I added it  (sensor_combined_.idl)

- This creates a basic subscriber and publisher, and a main-application to run them. To print out the data from the sensor_combined topic, modify the onNewDataMessage() method in sensor_combined_Subscriber.cxx:
```
void sensor_combined_Subscriber::SubListener::onNewDataMessage(Subscriber* sub)
{
    // Take data
    sensor_combined_ st;

    if(sub->takeNextData(&st, &m_info))
    {
        if(m_info.sampleKind == ALIVE)
        {
            // Print your structure data here.
            ++n_msg;
            std::cout << "\n\n\n\n\n\n\n\n\n\n";
            std::cout << "Sample received, count=" << n_msg << std::endl;
            std::cout << "=============================" << std::endl;
            std::cout << "gyro_rad: " << st.gyro_rad_().at(0);
            std::cout << ", " << st.gyro_rad_().at(1);
            std::cout << ", " << st.gyro_rad_().at(2) << std::endl;
            std::cout << "gyro_integral_dt: " << st.gyro_integral_dt_() << std::endl;
            std::cout << "accelerometer_timestamp_relative: " << st.accelerometer_timestamp_relative_() << std::endl;
            std::cout << "accelerometer_m_s2: " << st.accelerometer_m_s2_().at(0);
            std::cout << ", " << st.accelerometer_m_s2_().at(1);
            std::cout << ", " << st.accelerometer_m_s2_().at(2) << std::endl;
            std::cout << "accelerometer_integral_dt: " << st.accelerometer_integral_dt_() << std::endl;
        }
    }
}
```
I have made some changes in the above part, and the original text has a lot of underline

- To build and run the application on Linux:
```
$ make -f makefile_x64Linux2.6gcc
Checking directory output
Checking directory output/x64Linux2.6gcc
Checking directory lib
Checking directory lib/x64Linux2.6gcc
Checking directory bin
Checking directory bin/x64Linux2.6gcc
g++ -c -Wall -fpic -m64 -O2 -std=c++0x -I.   -c sensor_combined_Subscriber.cxx -o output/x64Linux2.6gcc/sensor_combined_Subscriber.o
g++ -m64 -o bin/x64Linux2.6gcc/sensor_combined_PublisherSubscriber output/x64Linux2.6gcc/sensor_combined_.o  output/x64Linux2.6gcc/sensor_combined_PubSubTypes.o output/x64Linux2.6gcc/sensor_combined_Publisher.o output/x64Linux2.6gcc/sensor_combined_Subscriber.o output/x64Linux2.6gcc/sensor_combined_PubSubMain.o -Wl,-Bstatic -Llib/x64Linux2.6gcc -lsensor_combined_  -Wl,-Bdynamic -lfastcdr -lfastrtps -lssl -lcrypto -ldl -lnsl -lm -lpthread -lrt
$
$ bin/x64Linux2.6gcc/sensor_combined_PublisherSubscriber subscriber
 Starting 
 Waiting for Data, press Enter to stop the Subscriber.

```

No data was received

TSC21, 2020-07-08:
Can you please post here what's the version of Fast-RTPS and Fast-RTPS-Gen you have installed. You should have minimal Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.4.

TSC21, 2020-07-08:
>ps: The last sentence of the document is missing an underline

That depends on the Fast-RTPS version you have installed.

Yoke365, 2020-07-08:
The versions are all correct，Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.4 . 　Reference https://dev.px4.io/master/en/setup/fast-rtps-installation.html

Yoke365, 2020-07-08:
> > ps：文档的最后一句缺少下划线
> 
> 这取决于您安装的Fast-RTPS版本。

The versions are all correct，Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.4 . 　Reference https://dev.px4.io/master/en/setup/fast-rtps-installation.html

Yoke365, 2020-07-08:
> > ps: The last sentence of the document is missing an underline
> 
> That depends on the Fast-RTPS version you have installed.

The versions are all correct，Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.4 . 　Reference https://dev.px4.io/master/en/setup/fast-rtps-installation.html

TSC21, 2020-07-08:
OK. So when you start the Listener, after starting the agent, do you get on the agent console "Publisher matched"?

Yoke365, 2020-07-09:
> OK. So when you start the Listener, after starting the agent, do you get on the agent console "Publisher matched"?
Ｉ try start the agent, after starting Listener.  still can't get the data

[image]

Yoke365, 2020-07-09:
> OK. So when you start the Listener, after starting the agent, do you get on the agent console "Publisher matched"?

[image]

Yoke365, 2020-07-09:
> OK. So when you start the Listener, after starting the agent, do you get on the agent console "Publisher matched"?

I didn't see the print of "publisher matched". I pasted the screenshot to help me have a look

lijieamd, 2020-12-15:
I had the same problem and i found that is caused by rtps version, you can try to remove all rtps and rtpsgen lib and install again
hope this may help you

mrpollo, 2020-12-15:
Thanks @lijieamd.

since this issue hasn't been updated in a while, I'm closing.

closed 2020-12-15 by mrpollo