PX4/PX4-Autopilot #15287: pwm_input driver will freeze FMU when the input signal is in uncertain level

OskarPepryk, 2020-07-08:
**Describe the bug**
FMU freezes when I plug in an unpowered mb12xx sensor to the AUX5 port and start pwm_input driver. When measured, the voltage on PWM pin is around 2V, for some reason. Plugging the power to the sensor unfreezes the board, as the signal is now correct at 0 or 5V. Tested on Pixhawk Cube with latest fmu-v3_default. Maxbotix sonar connected with pulse width pin. 

**To Reproduce**
1. Plug in Maxbotix sonar sensor not with ADC, but with PW pin connected to AUX5 signal pin.
2. Leave +5V of the sensor floating.
3. Start pwm_input driver
4. The autopilot will (not always) freeze as long as the power pin is unconnected. It will resume when reconnected.

**Expected behavior**
A connection break of the sensor, or having an uncertain logic level at PWM input pin should not freeze the FMU.

**Log Files and Screenshots**
N/A

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
Tested with a proof-of-concept setup, no actual vehicle.
**Additional context**
After resuming the operation, error_count field in pwm_input topic is showing large numbers. This suggests that the PWM interrupt triggers continuously freezing the FMU. Proposed solution is to stop the driver from within pwm_input ISR if the error count rises rapidly.

wattsinnovations, 2020-07-08:
>When measured, the voltage on PWM pin is around 2V, for some reason

How did you measure the voltage? With an oscilliscope? Could you provide a screenshot of the measured waveform?

dakejahl, 2020-07-08:
>When measured, the voltage on PWM pin is around 2V, for some reason

How did you measure the voltage? With an oscilloscope? Could you provide a screenshot of the measured waveform?

dakejahl, 2020-07-08:
I'm guessing there's a noisy waveform on the input pin, causing rapid triggering of the ISR -- possibly starving the rest of the system. The over capture flag is set, and the error count is being incremented https://github.com/PX4/Firmware/blob/master/src/drivers/pwm_input/pwm_input.cpp#L143-L147

dakejahl, 2020-07-09:
Give this a try. This is just a quick fix that will disable the IRQ if there are 3 consecutive over capture errors. I don't have the mb12xx hardware to test, so let me know if this fixes the issue.

https://github.com/PX4/Firmware/compare/pr-pwm_input_fix?expand=1

OskarPepryk, 2020-07-15:
> How did you measure the voltage? With an oscilloscope? Could you provide a screenshot of the measured waveform?

Yes, with an oscilloscope but it didn't record much noise that could trigger an interrupt. But that could be because I didn't look hard enough.

@dakejahl The fix works, the driver stops sending pwm_input messages and FMU keeps running. Driver could periodically try to restart the interrupt though - right now it is needed to do it manually.

dakejahl, 2020-07-15:
>Driver could periodically try to restart the interrupt though - right now it is necessary to do it manually.

As it stands the driver is run purely on interrupts. We could add rescheduling behavior for enabling the ISR, take a look at the work items available in **platforms/common/work_queue**

closed 2025-05-05 by dakejahl