PX4/PX4-Autopilot #15289: Unexpected flight termination

kuiskas, 2020-07-08:
**Describe the bug**
We were flying a mission with our coaxial octocopter, v1.10.1. Suddenly the motors were stopped and the copter fell from 120m altitude. I have been looking the reason at the logs but did not found anything yet. Could any body help here. Could be a problem with the firmware? 

**To Reproduce**
Nothing special. In mission mode all the motors stopped and automatically from mission mode, freefall not detected, to position mode. 

**Log Files and Screenshots**
The logs are available here.
https://review.px4.io/plot_app?log=44fcd17a-694d-431e-a02f-74c51f8d3cda

**Drone (please complete the following information):**
Coaxial Octocopter, Pixhawk 4, Here+ 2. 

**Additional context**
We implemented some changes on the original v1.10.1. We are testing a dual redundant smart battery. Therefore we have changed the original smbus driver. 
Anyway the copter has been flying around some time with this battery, I really do not know what happened here.

bkueng, 2020-07-09:
Sorry to read that.
The vehicle did disarm in-flight, while still in mission (shortly before switching to position mode), but I do not see why. It's not RC, not landed and no vehicle command.

kuiskas, 2020-07-09:
Hi @bkueng,
thanks for your help. It saw in the logs, the vehicle enter in the maybe landed and landed condition but no frefall detected.
I had several ideas but none of them could explain this. 
1. CPU overloaded, could this be? There is a peak just before everything happened. 
2. Could be vibrations?

kuiskas, 2020-07-09:
Could be this?
https://github.com/PX4/Firmware/issues/11683

MaEtUgR, 2020-07-09:
Are all those CUV or apparently "undervoltage protection" warnings expected?
It's first disarming, then detecting landed
[image]

MaEtUgR, 2020-07-09:
Why is there no disarm message? 🤔 :
[image]

MaEtUgR, 2020-07-09:
@kuiskas Do your changes contain anything in commander and maybe `arming_state_transition(... , vehicle_status_s::ARMING_STATE_STANDBY, ...);` ? Just to rule it out.

kuiskas, 2020-07-09:
Hi thank you all for you help here.
here there is a log from 2 days before the crash, with the same setup:
https://review.px4.io/plot_app?log=77a28b5c-f171-4f2c-8fc3-a98fa2beadf6
Later that day I flew a mission without problems. But I had the SD_LOG parameter set to from boot until disarm. 

The CUV it is expected, it is a glitch in the smbus communication.  It means cell undervoltage detected.
From Commander we did not changed anything.
Normally when landed and disarm we just get:
[image]

kuiskas, 2020-07-09:
Another thing we changed. The max number  of task in nuttx. 
In our setup we were using 4 mavlink instances. USB instance + 3 for telemetry and peripherie. So we increased the max number of task to 64 as in the newer firmware.
We did this because neither mavlink shell console nor the logger were working when this nunmber was 32 task. We had in total 33 task running.

kuiskas, 2020-07-10:
Could the logger be wrong and the disarm occurred after the landing state? 
Any clue why is not detected a freefall?